FastLED/FastLED #575: RGBW How to remap colors? Arduino Uno - SK6812/WS2812B GRBW or WWA

bogus105, 2018-03-03:
Hi!

I'd love to make some cool lamp using fire flame fliskering effect. Tried the 'fire2102' and 'palette' version examples from FastLED library with my LED strip - GRBW WS2812B (or SK6812?) diodes 60 per meter. 
Apparently an addition of one more color to the chip need one more color information in data packet:) 
Now all i can get is nice flickering effect but colors are out of real flames colors palette. 
Is there any way to define GRBW type leds? The only types compilator passed through were RGB, GRB, etc. If i try to add W it won't go. 
The only LED strips i have are Green-Red-Blue-White and White-White-Amber. I'm trying to make flame effect on both.
Thanks for inputs!

marmilicious, 2018-03-03:
RGBW leds are not supported yet.  You should be able to use your White, Warm White, Amber strip though and code the W,W,A while thinking in terms of R,G,B.   CRGB(0,0,255) would give Amber full brightness, CRGB(128,128,128) would give 50% brightness of each, etc.

bogus105, 2018-03-11:
Thank you for advice.

messismore, 2018-06-07:
See #482.