FastLED/FastLED #579: So good library but where is RGBW leds support?

bogus105, 2018-03-10:
Nowadays 4 chips ( Red, Green, Blue, White) LEDs are becoming so popular in lighting applications that RGBW leds support is a must. RGB leds are unable to light up your room with warm, nice looking light.
I've read some months ago that RGBW are not supported by FastLED and that this feature is on the list.
It was many months ago...

Is there any progress in this subject?

We need it so much:)

kriegsman, 2018-03-10:
RGBW (etc) support is at the top of our to-do list for our next big projects ... when we have time and energy to dig into it.  However, as you may or may not have read, Dan and I had a death in our family, and we're just not back up to speed yet on library development.

I hope you can find the tools you need for the projects you want to work on.  Right now, FastLED does not yet support RGBW.  We want it, too, but we're dealing with a lot of other things aside from the library.

Thanks for your support and for your "vote of interest" for RGBW.

closed 2018-03-10 by kriegsman

bogus105, 2018-03-11:
I am so sorry to hear that. Condolence.

Among other reasons i like FastLED lib the important one is "Flame2012" program example:) Would it be possible for you guys to write the description for this example for us, newbies?  I'd love to use this example to put flames on my WhiteWhiteAmber led strips. But firstly i'd love to UNDERSTAND the way you:
1. generate flame effect (that heat cells, etc.)
2. transform heat matrix to actual led colors

Once again: THANK YOU GUYS for a GREAT lib!:)

jaketesler, 2018-03-28:
+1 for support for RGBW (4-element) and RGBWW (5-element) :)

yesyesuk, 2018-03-29:
I'm also waiting for RGBW support before switching to this library from the Neopixel library.

But isn't RGBWW RGB plus Warm White? I have not seen 5 colour strips. Do you have a link to these?

jaketesler, 2018-03-30:
5 element strips: http://www.witop-tech.com/whats-difference-between-rgbw-and-rgbww-led-strip/

focalintent, 2018-03-30:
The re-write that I'm doing for FastLED is intended to address all of these options.  I'm making the underlying data structure for a pixel (currently CRGB - which is just an 8-bit RGB pixel) much more flexible, and by extension, everything that uses it (the code that writes led data out to pixels, color conversions, utility functions, etc...).  And by more flexible, I mean replacing it with something that can be configured as 8-bit RGB, or 8-bit RGBW, or 8-bit RGBWW, or 8-bit RGBWWW, or 8-bit RGBA (amber), really anything from 3-6 components.  Oh, and 16-bit versions of all of the above - which also means that you can use 16-bit data structures to write to 8-bit led controllers, or the reverse, using 8-bit data structures to write to 12 or 16-bit led controllers, etc... etc... etc...  

Which basically means the entire library is getting re-written :)

wyldbill, 2018-10-18:
@focalintent Any Updates?

focalintent, 2018-10-25:
When there are updates there will be posts both here and on g+

eshkrab, 2019-05-14:
I just finished an unofficial mod of FastLED to support RGBW (went low-level and tried to do more robust solution than https://www.partsnotincluded.com/programming/fastled-rgbw-neopixels-sk6812/)

kicker is I did this on top of my own port of FastLED from Arduino to esp-idf, but if someone who's actually on FastLED signs off on my approach, I'm happy to port the mod back to Arduino for the vanilla FastLED

In any case, my blunderings can be found at https://github.com/eshkrab/FastLED-idf-rgbw if anyone is curious

jaketesler, 2019-07-16:
@focalintent Any updates on RGBW+ support? :D

focalintent, 2019-07-16:
Not yet, no

DerDude87, 2019-09-24:
Hello, i want to ask about the progress with RGBW Support... Now it is more than a Year after ther first talk about ... and i hoped, now i can use Fast LED with an RGBW Stripe like WS2812B called SK8612... Im not so good in scripting ... a lot i understand but the most i dont ... The Strandtest from adalight is working ... also i make the script working that "eshkrab" create as example ... but its only a rainbowloop ... but i want to use it with Prismatik or some oter Ambyligtsoftware ... i hope that somebody can help me to make it work ... cause my old RGB Strip ist damaged an today i installed the new RGBW stripe ...   Thanks a lot for your help ... and sorry for my bad english ... Best Regards ...

kriegsman, 2019-09-24:
Dan and I were working on this. But as you may have heard, we lost Dan in a boat fire off the California coast three weeks ago. I’m grieving, along with the rest of the FastLED community, and the families. I want to support RGBW, but I don’t have any timeline.
Please follow discussion and announcements on the Reddit FastLED group. Thank you for your understanding in this extremely sad and difficult time.

DerDude87, 2019-09-24:
Im so sorry to read this. No i dont recieved this bad news! I hope u are fine. Take your Time and my most heartfelt condolences !!! Best Regards

BYohann, 2019-09-26:
Very sorry to hear about that, it's part of life we never can accept, my condolences to the family and friends. I hope he will find a lot of leds and lights where he is now

DerDude87, 2020-08-02:
Hi kriegsman, 
i hope, now u are better than last year. 

I would ask u for news... Is there hope to get a RGBW Support with SK8612 LED Strips an Fast LED with Prismatik ? 

Best regards -- Der Dude

jvanlangen, 2020-10-18:
So, any updates on this?

dmurph, 2021-02-02:
In case this is useful for anyone else, I experimented with some different rgb->rgbw transforms here:
https://www.dmurph.com/posts/2021/1/cabinet-light-3.html

The best one took into account the temperature of the white led:
https://gist.github.com/dmurph/c650904699d0b3a6db3dfefd8c2fded4#file-rgb2-ino

ChristofKaufmann, 2021-05-26:
@dmurph Thank you for your insights. I extended your approach a bit:

- brightness correction of white LED: My white LED was brighter than what I could achieve with the RGB LEDs at the same color.
- CRGBW inherits from CRGB: This allows to use a lot of CRGB functions and even a CRGBWArray to make life easier.

Here is a demo:
https://gist.github.com/ChristofKaufmann/71a6259a1483df180cde76cb032debe8