FastLED/FastLED #580: Inconsistent colors

lexam79, 2018-03-12:
Hi all,

I'm writing here too (I've also posted in BRUH's repository, whose project I'm trying to implement), in case someone is able to provide any useful feedback, as I'm starting to go crazy with this. So I built a project, with a WS8211 strip I got off of eBay and a Wemos D1 mini. The problem I'm seeing is that it seems as if leds get "out of sync", i.e. I turn it on (solid color) all leds are on. I turn it off, some leds stay on (at the end of the strip usually, maybe even switch to a different color - usually green). I turn it on, then off again, a different number of leds stay on, possibly with a different color... It's completely inconsistent. Sometimes, if I turn on/off several times, I might get all leds to turn off.

I've tried several things to sort it out, both hardware and software related, to no avail.

Can anyone shed some light as to what the problem might be?

Thank you

yesyesuk, 2018-03-12:
What exactly are you turning off and on?

lexam79, 2018-03-12:
The LED strip. I'm turning the LED strip on/off, via MQTT. This is the project I'm trying to implement: https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs

yesyesuk, 2018-03-12:
ah, OK. I thought you mean you turn power off and on. 
How long is the LED strip?

lexam79, 2018-03-12:
It's 4m, with 39 drivers and 3 leds per driver. The power supply is 8A (max consumption of the strip is 36W, according to seller) and I've even injected power in 3 places along the strip and added a 1000uF capacitor next to the Power Supply.

yesyesuk, 2018-03-12:
hmm, ok. injecting power along the strip is what I was trying to get at. ;-)
Then I'm out of ideas, I'm afraid...

lexam79, 2018-03-12:
I don't think it's power-related anyway, as it's stable. Inconsistent, but stable. I haven't seen any flickering, random turn offs or anything similar.

yesyesuk, 2018-03-12:
And it only happens when you turn the LEDs off (i.e. set them all to black)? Not when setting any other colour?

lexam79, 2018-03-12:
No, it also happens when I switch colors, change brightness etc. I just described the simplest issue, that of turning on-off.

focalintent, 2018-03-12:
You might be running into interrupt related problems - try adjusting the retry count, also see if turning off interrupts (just as a test) helps - https://github.com/FastLED/FastLED/wiki/Interrupt-problems

lexam79, 2018-03-13:
Hi focalintent, it seems that using `#define FASTLED_INTERRUPT_RETRY_COUNT 1` did the trick. I had tried that, as well as `#define FASTLED_ALLOW_INTERRUPTS 0` earlier, when I was trying to drive the strip directly off of the Wemos D1 (@ 3.3v), but did not try it again, after I placed a voltage shifter between the Wemos and the strip. Surprisingly, the behavior I was getting without the shifter (with or without the interrupt-related mods above) was identical to what I was getting after adding the shifter, but without the `#define FASTLED_INTERRUPT_RETRY_COUNT 1`. Anyway, it seems it's working now, so I'm happy and I'll live to go crazy a bit later... Thanks!

closed 2018-06-23 by focalintent