FastLED/FastLED #581: unable to compile with MicroCore for attiny13

c-e-p-x-u-o, 2018-03-21:
https://github.com/MCUdude/MicroCore/issues/54

```
In file included from /root/Arduino/libraries/FastLED/FastLED.h:51:0,
                 from /root/Arduino/RGB-HUE-RR/RGB-HUE-RR.ino:5:
/root/Arduino/libraries/FastLED/fastpin.h: In instantiation of 'class FastPin<1u>':
/root/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:96:49:   required from 'class ClocklessController<1u, 3, 3, 6, (EOrder)10u, 0, false, 10>'
/root/Arduino/libraries/FastLED/chipsets.h:460:7:   required from 'class WS2811Controller800Khz<1u, (EOrder)10u>'
/root/Arduino/libraries/FastLED/FastLED.h:111:52:   required from 'class WS2811<1u, (EOrder)10u>'
/root/Arduino/libraries/FastLED/FastLED.h:304:33:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2811; unsigned char DATA_PIN = 1u]'
/root/Arduino/RGB-HUE-RR/RGB-HUE-RR.ino:13:43:   required from here
/root/Arduino/libraries/FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^
exit status 1
Error compiling for board ATtiny13.
```

Code (it works with arduino atmega and Attiny85  ATTinyCore
```
// this is temporary workaround for ATTinyCore
extern "C" {
  volatile unsigned long timer0_millis = 0; // FAKE FAKE FAKE!
}

#include<FastLED.h>
#define NUM_LEDS 8

CRGBArray<NUM_LEDS> leds;

static uint8_t hue[NUM_LEDS];

void setup(){
  FastLED.addLeds<WS2811,1>(leds, NUM_LEDS);

  for(int i = 0; i < NUM_LEDS; i++){
    hue[i] = i;
  }
}

void loop(){ 
  for(int i = 0; i < NUM_LEDS; i++){   
    leds[i] = CHSV(hue[i],255,255);
    FastLED.delay(0);
    hue[i] ++;
  }
}
```

focalintent, 2018-03-21:
The problem is there are no pin definitions for the ATTiny13 in https://github.com/FastLED/FastLED/blob/master/platforms/avr/fastpin_avr.h - you would have to add those, as the attiny13 isn't a supported configuration yet.

Sagar73594, 2022-08-10:
It's been almost 2 years, Still no support? @focalintent

marmilicious, 2022-08-10:
@Sagar73594  Unfortunately Dan (aka focalintent) is no longer with us.
https://laist.com/news/conception-boat-fire-santa-cruz-island

JulianGmp, 2023-01-22:
For those coming across this in the future, I tried porting over FastLED to the attiny, but I would say it's not really possible.

You can find some discussions on the [FastLED reddit](https://www.reddit.com/r/FastLED/comments/10i7cf7/is_porting_the_attiny13_even_feasible/), but the tldr is:
- the ATtiny13/ATtiny13A's [micro core](https://github.com/MCUdude/MicroCore) doesn't provide a ms counter, which FastLED (from my understanding) relies on.
- The chip is also limited to **1024 bytes** of program memory and 64 bytes of RAM. FastLED already gives out a warning to ATtiny25 users, that the 2048 bytes storage is likely not enough.

*For these reasons, I would say using FastLED on the ATtiny13 or ATtiny13A is not possible.*

---

For those who happen to have WS2812 LEDs: you may want to take a look at Tiny WS2812 ([platformio](https://registry.platformio.org/libraries/ctxz/Tiny%20WS2812), [github](https://github.com/CTXz/TinyWS2812)).

I managed to get that one running and did a simple color cycle for 6 LEDs. This already consumed 480 bytes of the program memory though. Adding any sort of complex functionality like HSV to RGB conversion will exceed the 1024 bytes limitation.
Usable for very, very simple applications, but if you're shopping for micro controllers, I'd say go for one with more storage, I'll probably buy some ATtiny85's.