FastLED/FastLED #582: PCA9685 Support?

johnyradio, 2018-03-22:
hi

can i control the PCA9685 from FastLED running on Arduino Uno or duemilanove?

thx

tonton81, 2018-03-22:
its a free running i2c pwm device, short answer is no. i2c is too slow and you dont control fastled with a consistant pwn

tonton81, 2018-03-22:
unless its normal RGB strip that dont have any chips inside them

focalintent, 2018-03-22:
The PCA9685 isn’t currently supported in FastLED.  You can use fastled’s color code to set up your led data, and then use your own code to write it out to the 9685, though.

johnyradio, 2018-03-22:
That's what I have in mind. 

But is that the best way to generate color patterns for the PCA9685?

Are there any other color pattern libs?

demlak, 2018-10-16:
I also want to use a PCA9685. Any progress on this?

focalintent, 2018-10-25:
Again - you can use FastLED to populate your CRGB array, and then write out your own RGB values to the PCA9685 using one of the PCA9685 libraries out there.  I'm unlikely to add support for the PCA9685 into FastLED until after the RGBW/RGB16 rewrite is finished (in part because the PCA9685 would want 12-bit color data).

johnyradio, 2018-10-25:
Thanks so much for all your help and hard work! Thx for FastLED!