FastLED/FastLED #583: First and last LED show wrong color

xcruell, 2018-03-24:
Hello,
i recently made a ceiling light with ws2811 leds, but theres one little problem..
The first and last LED show the wrong color, and not only the wrong one, they only show green and blue.
I made a video to demonstrate the issue. You can clearly see, that the first led turns blue, when blue is near the end of the strip. Its like it gets the color from the wrong spot, if that makes sense.

When i up the NUM_LEDS count by one, i can workaround the issue for the last led, because it would show up on the new non existing last one. But i cant do that with the first LED.

The code i used: https://pastebin.com/aSqQedi6
Video demonstration: https://youtu.be/PHB16HLMtdo
Im running the code on my arduino nano.
For some reason i cant run any FastLED sketch with more than ~73 leds on my esp8266/32s

I hope that someone can help me out :)

focalintent, 2018-03-24:
What version of the library are you using?  There was an old off by one error in the ws2812 output code on avr - but that was a long time ago.  Also is the nano you're using running at 5v or 3.3v?  I have sometimes seen issues with WS2811's where some data gets lost, usually related to poor level shifting.

As for the esp8266/32 - have you tried adjusting whether or not interrupts are allowed or the retry count - https://github.com/FastLED/FastLED/wiki/Interrupt-problems

xcruell, 2018-03-24:
Hey!
Im using 3.1.6. My Nano is running at 5v.
Maybe thats the issue with the esp8266, since the digital pins output 3.3v. 

I tried both, allow interrupts and retry count without success on the esp.

peto2006, 2018-10-03:
I can confirm that esp8266s pins output 3.3V. With my WS2812B, it sometimes worked fine, but sometimes LED strip couldn't figure where byte stream starts and all colors got shifted by one. Buying Logic level converter from 3.3V to 5V solved this particular issue for me. (I still have other problems with esp8266 and WS2812B.)

Blink515, 2018-10-05:
> Im actually using a level shifter now.
> I had to because i got flickering problems with my 235 WS2811 and with 70+
> of WS2812B.
> But the shifted start still remails. Isnt there a way to "sync" the first
> and last LED ?

Can the first and last LEDs show the full spectrum of colors? Maybe one of
the individual red, green, or blue LEDs on the die are dead?

xcruell, 2018-10-05:
Yes they can.
I got them on white atm, no difference what so ever.
The main problem really is the "shifted start point".
The last LED is basically the first one showing the pattern. 
But the first LED is the first wired up to the esp.
I dont have this problem with other libraries, only FastLED.

focalintent, 2018-10-05:
3.1.6 is fairly old at this point - you should grab the most recent version and see if that fixes the problem.  Also your pastebin has expired so I can’t see what the code is that you have.

xcruell, 2018-10-05:
Im using 3.2.1 and a different code by now.
https://pastebin.com/LD4j48CA

regards