FastLED/FastLED #586: ESP32 and Smartmatrix HUB75 - Needs DMAChannel.h ?

winini, 2018-03-29:
Hello,

I am trying to use a RGB matrix (HUB75) with FastLED.
I installed the Smartmatrix library, but I can not compile the code.
DMAChannel.h error missing.
This file seems to be part of teensy core.

D:\Documents_D\Arduino\libraries\SmartMatrix\SmartMatrix.cpp:26:24: fatal error: DMAChannel.h: No such file or directory
Erreur de compilation pour la carte ESP32 Dev Module
I tried with Smartmatrix 3.x and 2.x but it's the same message.
I tried to comment the lines related to the DMA, but it's worse.

This RGB matrix works well with various ESP32 and adafruit_gfx, but I'd love to get there with FastLED.
I have not yet figured out in which file you can define the GPIO numbers connected to the HUB75, unless it is automatic?
But it's not the most important thing, it should first that the compilation works for my ESP32.
Thanks for your help.

focalintent, 2018-03-29:
The SmartMatrix library is for a specific RGB led matrix, and my understanding is the library requires the Teensy 3.x.  You would need to first port this library - https://hackaday.com/tag/hub75/ - to the esp32 - and then you could write a wrapper for that library for FastLED like the SmartMatrix wrapper in FastLED.  

Your compilation errors have to do with the SmartMatrix library, not FastLED.

closed 2018-03-29 by focalintent

winini, 2018-03-29:
Yes I understand.
I also left my question on the Smartmatrix page.
This is in case someone has already worked on this port to ESP32.
I'm going to watch closely but it seems complicated for me.

Thank you for your response, and your work.

focalintent, 2018-03-29:
For a general question like this for the wider community, try asking on the g+ group - FastLED.io/+

winini, 2018-03-29:
ok thanks for the information.