FastLED/FastLED #587: Request for calculate_unscaled_power_mW(led_buffer, num_leds) and such to be made into public functions for CFastLED

connornishijima, 2018-03-29:
I'd really like to see something like this be a possibility in the next update:

Let's say our LEDs are first set in a way that would draw exactly 1A, but then I use a "setMaxPower" function to enforce software limiting to 750mA:

    FastLED.setMaxPowerInVoltsAndMilliamps(5, 750);

I wish there were a feature to pull current data about the software power management without setting #defines inside the library that would be reset upon updating it. For example:
    
    FastLED.get_power_raw_mA(leds, NUM_LEDS, volts);          // = 1000 (mA)
    FastLED.get_power_limited_mA(leds, NUM_LEDS, volts);      // =  750 (mA)
    FastLED.get_power_overdraw_mA(leds, NUM_LEDS, volts);     // =  250 (mA)
    FastLED.get_power_dim_level(leds, NUM_LEDS);              // =  0.75
    FastLED.get_power_dimming(leds, NUM_LEDS);                // =  true
    
Now this works:

    if( FastLED.get_power_dimming(leds, NUM_LEDS) ){ // true

      // Prints "Power overdrawn by 250mA"
      Serial.print("Power overdrawn by ");
      Serial.print( FastLED.get_power_overdraw_mA(leds, NUM_LEDS, 5) );
      Serial.println("mA");

      // Prints "Dimmed to 75.00%"    
      Serial.print("Dimmed to ");
      Serial.print( FastLED.get_power_dim_level(leds, NUM_LEDS)*100 );
      Serial.println('%');
    }

the-magister, 2018-03-29:
Connor,

You've described the "what" of the ask very precisely.  Could you also
elaborate the "why" behind the ask?  In the context of competing priorities
and limited resources, that information is important.

Put bluntly, are there circumstances in your use case that cannot be solved
by simply selecting an adequate power supply?  I ask, as there's overhead
associated with counting up all of the power requirements, so the current
"simple" system isn't in any way "free".  Additional complexity is even
less free.  Are you sure you want the codebase to run more slowly for this
feature?

Mike

connornishijima, 2018-04-18:
I sell edge-lit numeric displays [**like this**](https://www.youtube.com/watch?v=DaQvmIobrKs) that are designed to be easy for beginners - and the Arduino library for them relies on FastLED as a dependency. Because a chain of 6 numbers (20 WS2812B ea.) can _attempt_ to draw up to ~7.2A if written to irresponsibly, I'm trying to add functions to monitor the approximate power usage your LED animation consumes at any given time. This way my users could get a live idea of their power supply requirements, and be warned if a particular effect is overstepping their current setup. (Again, many of them are new to electronics)

Correct me if I'm wrong, but I don't think setting the existing private function _**calculate_unscaled_power_mW(led_buffer, num_leds)**_ as a public function instead would slow anything down. It already exists, it just can't be read from outside of the FastLED library. And if the user never calls a library function, it isn't even compiled in, so it wouldn't hurt those that don't need it.

And sure, most people don't need it. But then again most people need sfract15, CHSVPalette32, or other existing niche FastLED features for their projects. They're just nice to have when you _do_ need them! :)

focalintent, 2018-06-23:
Look at the functions in https://github.com/FastLED/FastLED/blob/master/power_mgt.h -- those should all still be public/available (in fact, those are the functions that the FastLED object uses under the hood to do its work)

closed 2018-06-23 by focalintent