FastLED/FastLED #588: wrong color on a code for a rainbow

Hexxdroid, 2018-03-29:
hello, i just bought an addresable ledstrip, and want to make a rainbow wave. im using arduino nano to controll and power the led.

i have tried different option such as neopixel and it still also have the same problem.

here is the code that im using: (i would appreciate it alot for helping me)

#include <FastLED.h>

#define PIN 4
#define NUM_LEDS 160

CRGBArray<NUM_LEDS> leds;
uint8_t hue[NUM_LEDS];

void setup() {
  FastLED.addLeds<NEOPIXEL, 4>(leds, 160);
    for (int i = 0; i < NUM_LEDS; i++) {
      hue[i] = 255 / NUM_LEDS*i;
      
      }
      
}

void loop() {  
  for (int i = 0; i < NUM_LEDS; i++) 
  {
    leds[i] = CHSV(hue[i]++, 255, 255);
  }
  
  FastLED.show();
  delay(20);
 
}

JosefKuchar, 2018-04-01:
First of all try RGB, if red is red, green is green, blue is blue. If not, you have different color order (it's GRB for me). Look at this https://github.com/FastLED/FastLED/wiki/Rgb-calibration

Hexxdroid, 2018-04-02:
the colors are correct , I have done that.

Hexxdroid, 2018-04-02:
I followed a guide to see wich chipset my led have. (there was some problem here)
but the colors where wrong and if i tried diffrent RGB ordering(1 red led, 2 green led & 3blue led) and it there was even more problem.

so i tried only 1 led at the time . I started with red
and my red ordering was blue si no proble there

the i do the 2 next and there was more problem
[image]
the led should be  1 red en 2 green

the i tried all of them again and i got this
[image]

some of the led wont even show up...

JosefKuchar, 2018-04-02:
I successfuly compiled my sketch with SK6812 declaration, you can try it

Hexxdroid, 2018-04-02:
I have RGBW led, is this like the problem of it?

Hexxdroid, 2018-04-02:
@JosefKuchar I don't understand you. did you send code?

JosefKuchar, 2018-04-02:
@Hexxdroid Looks like RGBW strips are not supported yet #482

Hexxdroid, 2018-04-02:
well thank you very much for helping me.
I was at a lost and you have brght me up thank you! you have my eternal gratitude!

closed 2018-06-23 by focalintent