FastLED/FastLED #589: Support for SJ1221/UCS5603

JimmyM01, 2018-03-31:
I've been trying to get some new LEDs working. They're SJ1221s at Shiji-led. They're 12v with 3 dies per color. The SJ1221s have adjustable constant current and are fault tolerant with dual data lines. However their protocol doesn't jive with common WS2811/2812 LEDs. Looking at the protocol it seems that each color in a frame is 12-bits. 8 bits for the brightness of the color and 4 bits to set the constant current drive
I've attached the protocol and datasheet if you want to take a look.
[SJ1221 control protocol.pdf](https://github.com/FastLED/FastLED/files/1865669/SJ1221.control.protocol.pdf)
[SJ1221 IC CHIP DATASHEET.pdf](https://github.com/FastLED/FastLED/files/1865670/SJ1221.IC.CHIP.DATASHEET.pdf)

Even though the constant current bits are allowed per LED, just setting it like setBrightness() would probably be sufficient instead of inserting it the led array. So CRGB could still stay at 3x8bits.
What sort of effort is involved in creating a controller type in FastLED to output 12bits per color instead of 8? 
These strips are quite new but due to their high brightness and 12V compatibility, I'm hoping to see more of them in the wild.

JimmyM01, 2018-04-01:
That protocol is confusing me more and more. I just want some code that I can use to queue up data and send it out with WS2812 timing. WTF is a work code!? Some parts lead me to believe that each led just needs 12bits (8color+4current) but then there's some work code business and then greydata vs grey code. I'm sure part of it is the bad english. But I need to test.

JimmyM01, 2019-01-20:
> This appears to be a 12-bit chipset, I'll look into adding this once the RGBW/16-bit support work is done.

Thanks, Daniel

jiwaniaziz, 2020-05-18:
Hi, any update on adding support for SJ1221/UCS5603?

JimmyM01, 2020-05-18:
Shiji-led has discontinued the SJ1221 and SJ1211s.

jiwaniaziz, 2020-05-18:
Noted. UCS5603 is in mass production.

JimmyM01, 2020-05-18:
Do you have links to devices?

jiwaniaziz, 2020-05-18:
Noted. 
UCS5603 is in mass production, datasheet attached for reference if anyone wants to take a look. Is support available for any other 12-bit data packet that we can refer to. 
[UCS5603控制协议 (1).pdf](https://github.com/FastLED/FastLED/files/4645415/UCS5603.1.pdf)
[UCS5603  2017-3-31-英文.pdf](https://github.com/FastLED/FastLED/files/4645417/UCS5603.2017-3-31-.pdf)

jiwaniaziz, 2020-06-08:
Did anyone get a chance to look into UCS5603?

rudcode, 2023-05-25:
Here is a link to my implementation using esp32 and rmt driver [https://github.com/rudcode/platformio_esp32_ucs5603_test/blob/master/src/main.cpp](https://github.com/rudcode/platformio_esp32_ucs5603_test/blob/master/src/main.cpp)

Maybe this can help.