FastLED/FastLED #590: Fastled STM32 support

janWeinands, 2018-04-01:
Dear all,

How can we use fastled with arduino IDE and STM32F103RB nucleo board or maple mini?
I tried several forks from the library, several boards implementation, but with every implementation i get different errors. 
What is the correct way of using the STM32 boards with Fastled?

Thank you in advance.

jordanapplewhite, 2018-04-02:
Hello! You probably want to refer to the google FastLED Users group for support:
https://plus.google.com/communities/109127054924227823508

I was able to get FastLED compiling for Maple Mini with Kolyshkin's fork here
https://github.com/FastLED/FastLED/pull/568

janWeinands, 2018-04-04:
I also tried that one, But Where can i place the reg file he mentioned? I Couldnt find a CMSIS directory in the IDE. Thank you in advance.

marcmerlin, 2018-04-15:
people who may help you with support questions do not look at issues, which are meant to be used for bugs. Please use the G+ community.

Fxlr8, 2018-12-09:
@janWeinands, have you solved this problem?