FastLED/FastLED #591: ESP 32 resetting [Core  0 panic'ed (Interrupt wdt timeout on CPU1)]

Polishfuze, 2018-04-03:
I've made some code to control the lights in my room via WiFi and it worked fine on the ESP32 like a month ago, then I changed over to an ESP8266 (cheaper and smaller) which also worked fine, but I wanted to use both cores. When I uploaded my code I got the following message in the serial monitor:
```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU1)
Register dump:
PC      : 0x400d2f7e  PS      : 0x00060734  A0      : 0x80085555  A1      : 0x3ffc76f0  
A2      : 0x00000008  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffc7d28  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc32fc  A9      : 0x3ffc32e0  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x3ffc73a0  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

```
Backtrace: 0x400d2f7e:0x3ffc76f0 0x40085552:0x3ffc7710`
I've reduced my code and know for a fact neither the WiFi nor using both cores is a problem (by trial and error) so my fully reduced code that still spits out the same error is:
```
#include "FastLED.h"

#define NUM_LEDS 900
#define DATA_PIN 2

CRGB leds[NUM_LEDS];

void setup() {
  // put your setup code here, to run once:
  FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);
}

CHSV color = CHSV(10, 0, 0);
void loop() {
  delay(1000);
  fill_solid(leds, NUM_LEDS, color);
  FastLED.show();
}
```

keesdekoning, 2018-04-05:
Put this first in line #define FASTLED_ALLOW_INTERRUPTS 0
Even before #include "fastled.h"
The reason is that the other core has not been stopt ore shut down.
I had the same problem

Polishfuze, 2018-04-15:
Thanks the first solution worked, but I don't know how to close the issue

marcmerlin, 2018-04-15:
next to the comment green button, you have close issue

closed 2018-04-15 by Polishfuze