FastLED/FastLED #595: NRF52 status

vshymanskyy, 2018-04-12:
NRF52 is listed in the upcoming platforms, NRF51 looks to be included already.
Just wanted to create this issue to track the progress. How complex is this, provided that NRF51 support is available? Thx!

focalintent, 2018-04-13:
My memory from looking at the data sheets a couple years ago was that there were some significant changes, so it wasn’t clear how much of the nrf51 work I would be able to reuse.  At the moment I’m not doing any major work on the library still, and I’m not aware of anyone attempting an nrf52 port yet.

nickpleis, 2018-07-23:
Just curious as to if anyone is sitting on any progress here?

baramuse, 2018-09-04:
Same here !!
I'll try to dig a little bit to understand the scope of work !

nickpleis, 2018-09-04:
I’d be interested in helping. Are there any docs that might be a good
starting point?

baramuse, 2018-09-10:
Well we have the [fastLED implementation of the nrf51](https://github.com/FastLED/FastLED/tree/master/platforms/arm/nrf51), and we have the working [nrf52 on the Adafruit neopixel library](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.cpp#L1206).
I never been that low level on cpp programming but I can try figuring things out from those two !
@pcaddict would you share your discovery so far ?

pcaddict, 2018-11-30:
You can find my fork of FastLED [here](https://github.com/pcaddict/FastLED). This is by no means a working solution at this point. Right now I have only begun working on SPI based LEDs as that is what I have to test with. Colors and strip length are pretty much ignored. The FirstLight example will walk a white-ish color down the strip but all of the pixels that are supposed to be dark are red or green in color.

I am using an Adafruit NRF52 Feather with the latest BSP. The changes I made might not work with a BSP older than 0.9.0.

henrygab, 2019-08-03:
@vshymanskyy nRF52 chipsets now generally work, with three outstanding issues:

1. Timing issue causes eventual lockup on longer clockless LED strands (workaround exists, not pushed to master yet) -- See #840 for updates

2. First call to showPixel() causes incorrect first LED color for clockless LEDs (cause not determined yet) -- See #856 for updates

3. Maximum strand length hard-coded in library (workaround is to change the header to a larger maximum when you update your library … it's a balance between memory use and broad support, currently set to 144)

henrygab, 2019-10-11:
FYI, this issue should now be closed.  The NRF52 is supported.

closed 2019-10-11 by kriegsman

vshymanskyy, 2019-10-11:
Thanks everyone!