FastLED/FastLED #598: Atmega32 support(mighty core)

Master811129, 2018-04-16:
hi guy,  i have an issue with this library. when i coding with this lib it doesn't compile.
error: multiple library were found.

and when i change the model to arduoino-NG (atmega 8) it works very well.
please help

focalintent, 2018-04-17:
What are you building it with, what version of the library are you using, what does the entire build output look like? 

error: multiple library were found makes it sounds like you have multiple versions of FastLED installed.

I'm not aware of anyone using the mighty core - so it's likely that the library is missing pin definitions for your board (see platforms/avr/fastpin_avr.h) - but your error message isn't anything that comes out of FastLED itself, so I think you have some other build problems going on as well.

obdevel, 2018-07-01:
I'd welcome support for MightyCore too. Currently it results in a compilation error:
```
In file included from /Users/xxx/Documents/Arduino/libraries/FastLED/FastLED.h:51:0,
                 from /Users/xxx/Documents/Arduino/xyz_test_v1/xyz_test_v1.ino:8:
/Users/xxx/Documents/Arduino/libraries/FastLED/fastpin.h: In instantiation of 'class FastPin<13u>':
/Users/xxx/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:96:49:   required from 'class ClocklessController<13u, 4, 10, 6, (EOrder)66u, 0, false, 10>'
/Users/xxx/Documents/Arduino/libraries/FastLED/chipsets.h:382:7:   required from 'class WS2812Controller800Khz<13u, (EOrder)66u>'
/Users/xxx/Documents/Arduino/libraries/FastLED/FastLED.h:94:34:   required from 'class NEOPIXEL<13u>'
/Users/xxx/Documents/Arduino/libraries/FastLED/FastLED.h:310:28:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = NEOPIXEL; unsigned char DATA_PIN = 13u]'
/Users/xxx/Documents/Arduino/xyz_test_v1/xyz_test_v1.ino:90:53:   required from here
/Users/xxx/Documents/Arduino/libraries/FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
...
...
exit status 1
Error compiling for board ATmega644.
```
Arduino 1.8.5, macOS 10.13.5, 644P, MightyCore 2.0.0, standard pinout.
Attempting to use pin PD5, which maps to Arduino pin 13. This pin has no special function on this board, other than being PWM-capable.

Happy to help out with the port if there is documentation available.

Thanks

focalintent, 2018-07-05:
Again, pin definitions need to be added in platforms/avr/fastpin_avr.h for this board.

obdevel, 2018-07-09:
Ok, the 644P seems similar enough to the 1284P which is already supported. From my *limited* testing, I can compile and successfully run trivial tests by adding the 644 to the 1284 line in platforms/avr/fastpin_avr.h, e.g.

```#elif defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644P__)```

Maybe this will provide support for all the MightyCore supported MCUs. @Master811129 - do you want to try this ?

Are there any acceptance tests that can be run ? I'm using an own-design board so I don't have access to every pin to test.

tmbomber383, 2018-09-08:
I use 644P processors as well. I'm in work on Saturday and tried using this library for the first time and ran into this exact, same, issue...
and came up with the exact, same, solution.
In fastpin_avr.h I modified line 183 from:
#elif defined(__AVR_ATmega1284P__)
to:
#elif defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644P__)
and I'm up and running.
So, confirmation on this being a solution and a second request that it be folded in to the next release.