FastLED/FastLED #599: Feastled does errors to Analog Input on ESP8266

bobybobybob, 2018-04-18:
Voltage measure is constant without Fastled on ESP8266  A0

Integrate fastled in code will fluctuate Voltage measure even with no LED is connected.

focalintent, 2018-04-18:
Without seeing the code or the way you have things wired it’s difficult to say for sure - but I wonder if analog reads require interrupts and for some led types FastLED dissembles interrupts

bobybobybob, 2018-04-18:
Ok give me one day and i post here a sample to show problem.

focalintent, 2018-04-18:
Also describe how you have things wired up.

bobybobybob, 2018-04-18:
Sorry, i can not reproduce problem with simple code. Result are the same with and not fastled. Look like problem is elsewhere maybe the Wifi ? https://github.com/esp8266/Arduino/issues/2070

```
void setup() {
Serial.begin(9600);
}

void loop(){
unsigned long now=millis();
Fastled(now);
readAnalog(now);
}

void Fastled(unsigned long now){
  static unsigned long lastReadTime;
  if(now-lastReadTime>=2000){
  lastReadTime=now;}}

void readAnalog(unsigned long now){
static unsigned long lastReadTime;
if(now-lastReadTime>=1000){
  float sensorValue = analogRead(0); 
  Serial.print("a0:  ");
  Serial.println(sensorValue);
  lastReadTime=now;
}}
```

```
#include "FastLED.h"
#define DATA_PIN D5
int ledState = LOW;
CRGB leds[1];

void setup() {
Serial.begin(9600);
FastLED.setBrightness(70);
FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, 1);
}

void loop(){
  unsigned long now=millis();
  Fastled(now);
  readAnalog(now);
}

int Fastled(unsigned long now)
{
  static unsigned long lastReadTime;
  if(now-lastReadTime>=2000)  {
     if (ledState == LOW) {
     leds[0] = CRGB::Green;
     FastLED.show();
     ledState = HIGH;} 
     
     else {
     leds[0] = CRGB::Red;
     FastLED.show();
     ledState = LOW;}
 lastReadTime=now;}}

void readAnalog(unsigned long now){
static unsigned long lastReadTime;
if(now-lastReadTime>=1000){
  float sensorValue = analogRead(0); 
  Serial.print("a0:  ");
  Serial.println(sensorValue);
  lastReadTime=now;}}

```

WeMos D1 Mini
Put 1kOhm from 3.3V to A0
APA106 on 3.3V and Data vom D5

closed 2018-04-18 by bobybobybob