FastLED/FastLED #602: Bug in ColorFromPalette for CHSVPalette16 and CHSVPalette32

schollman, 2018-04-21:
Hi,

I think there is a bug in ColorFromPalette when the hues decrease in value (go backwards)
The following code does the hue interpolation.

uint8_t deltaHue = (uint8_t)(hue2 - hue1);
if( deltaHue & 0x80 ) {
	// go backwards
	hue1 -= scale8( 255 - deltaHue, f2);
} else {
	// go forwards
	hue1 += scale8( deltaHue, f2);
}

The first parameter to scale8 in the backwards case should be equal to the abs(deltaHue) but as coded is equal to abs(deltaHue) - 1. As a result, the value of hue2 + 1 is never returned.

For a concrete example, imagine the two cases 
1) hue1 = 0, hue2 = 10
	deltaHue = 10
2) hue1 = 10, hue2 = 0
	deltaHue = -10 (246)

In 1), the forward case, the first parameter to scale8 will be 10, returning values in the range 0 to 9 as desired.

In 2), the backwards case, the first parameter to scale8 will be 9 (255-246), returning values in the range 0 to 8, one short of desired.

kriegsman, 2018-05-07:
Yep, I think this is a bug; I'll get on it.

kriegsman, 2018-05-17:
Thanks for the clear bug report; I believe that the change that I just committed helps solve this problem. Please do let me know if it solves it for you, or if you're seeing further issues, and we'll reopen the ticket if needed. Thanks again.

commit ade8254a (2018-05-17): Correct an off-by-one bug in hue calculation.  Hopefully fixes #602.
commit dbd2d8b8 (2018-05-17): Correct an off-by-one bug in hue calculation, part II.  Hopefully fixes #602 the rest of the way.
closed 2018-05-17 by kriegsman