FastLED/FastLED #606: Archiving built core (caching)

pfmartins4, 2018-05-11:
I'm trying to use the library in a genuine arduino mega2560. And the same error keep going on. Not even the exemples are running.

Build options changed, rebuilding all
In file included from C:\Program Files (x86)\Arduino\libraries\FastLED\examples\ColorPalette\ColorPalette.ino:1:0:

C:\Program Files (x86)\Arduino\libraries\FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.008

 #    pragma message "FastLED version 3.001.008"

                     ^

Sketch uses 5372 bytes (2%) of program storage space. Maximum is 253952 bytes.
Global variables use 302 bytes (3%) of dynamic memory, leaving 7890 bytes for local variables. Maximum is 8192 bytes.

thanks

NicoHood, 2018-05-11:
This is not an error, just a message?

pfmartins4, 2020-04-26:
I guess somehow it's a  common mistake.