FastLED/FastLED #610: LED gone MAD

Shantoun, 2018-06-09:
So I want to control certain LEDs (in this case the first turning on red), so I used this code after downloading this library, hooking up the live and ground to a 220v to 5v converter and the data feed into the arduino.

#include <FastLED.h>

#define LED_PIN     7
#define NUM_LEDS    30
CRGB leds[NUM_LEDS];

void setup() {
  FastLED.addLeds<WS2812B, LED_PIN, GRB>(leds, NUM_LEDS);
  
}
void loop() {
  
  leds[0] = CRGB(255, 0, 0);
  FastLED.show();
  delay(500);  
  
}

The problem is when I plug in the LEDs the first 8 or so turn on full. Moreover, the LEDs are going crazy and flashing all sorts of colors and patterns. Please any thoughts?

Blink515, 2018-06-09:
You need to connect your ground to the Arduino as well.

Shantoun, 2018-06-10:
Thank you, that fixed everything!!

closed 2018-06-12 by focalintent