FastLED/FastLED #614: Color issues with SparkFun Lumenati

jd-boyd, 2018-06-14:
The product description says these boards use APA102C LED chips.  See here: https://learn.sparkfun.com/tutorials/lumenati-hookup-guide

I'm finding that for colors CRGB:Green works correctly, but CRGB::Blue gives me red and CRGB::Red gives me blue.

I will dig into it further when I can back to it, but I wanted to note the problem in case anyone else knew anything about it already.

makarij, 2018-06-14:
#define COLOR_ORDER GRB - whatever sequence RGB try to swap mismatching ones. For me worked:)

marmilicious, 2018-06-14:
@jd-boyd  You should be able to sort out what you need to use with the FastLED example RGBCalibrate.ino here:
https://github.com/FastLED/FastLED/blob/master/examples/RGBCalibrate/RGBCalibrate.ino