FastLED/FastLED #617: esp32 apa102 at GPIO33 and GPIO32 not working

MarkusBec, 2018-06-22:
i think there is something missing:

/lib/FastLED_ID126/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
#      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
^
In file included from /lib/FastLED_ID126/FastLED.h:51:0,
from src/main.cpp:24:
/lib/FastLED_ID126/fastpin.h: In instantiation of 'class FastPin<32u>':

/lib/FastLED_ID126/fastspi_bitbang.h:22:49:   required from 'class AVRSoftwareSPIOutput<32u, 33u, 20u>'
/FastLED_ID126/fastspi.h:30:7:   required from 'class SPIOutput<32u, 33u, 20u>'
/lib/FastLED_ID126/chipsets.h:94:6:   required from 'class LPD8806Controller<32u, 33u, (EOrder)136u, 20u>'
/lib/FastLED_ID126/FastLED.h:250:77:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = (ESPIChipsets)5u; u
nsigned char DATA_PIN = 32u; unsigned char CLOCK_PIN = 33u; EOrder RGB_ORDER = (EOrder)136u]'
src/main.cpp:70:69:   required from here
/lib/FastLED_ID126/fastpin.h:207:2: error: static assertion failed: Invalid pin specified
static_assert(validpin(), "Invalid pin specified");

GPIO33 and 32 should be regular GPIOs but they are out of the 32bit range.

focalintent, 2018-06-23:
It has less to do with 32-bit range and more to do with they are accessed via different GPIO registers - and so only pin definitions up to pin 31 are specified in https://github.com/FastLED/FastLED/blob/master/platforms/esp/32/fastpin_esp32.h 

Something along the lines of what was done for esp8266 - https://github.com/FastLED/FastLED/blob/master/platforms/esp/8266/fastpin_esp8266.h - or some such.

@samguyer - you have a comment in there about needing special handling - do you know what handling it'd want?

samguyer, 2018-06-23:
@focalintent The issue is that the FastPin definition that I inherited in fastpin_esp32.h uses a 32-bit value for the bit mask, so it cannot represent pins 32 and 33. However, I don't use the code in FastPin at all. Is there any reason to keep it? Could I just define the FastPin class as an empty template?

focalintent, 2018-06-23:
So there's the way I did it for the esp8266 - https://github.com/FastLED/FastLED/blob/master/platforms/esp/8266/fastpin_esp8266.h where pin16 is checked for with a constant check at compile time - but the better way is probably something along the lines of how it's done on most other platforms, where an instantiation of FastPIN involves both a pin number _and_ a base port register - which does get a little bit dicier to set up.  

You could probably make an _ESPPIN_HIGH for the above 31 pins and just use a different variation of the DEFPIN macro for pins 32 and 33 that references the _ESPPIN_HIGH template instead of _ESPPIN.

samguyer, 2018-06-23:
In the ESP32 case, I never directly write to any of the ports/pins. Given that, could I get away with defining FastPin like this...

```
template<uint8_t PIN, uint32_t MASK> class _ESP32PIN { };
#define _DEFPIN_ESP32(PIN) template<> class FastPin<PIN> : public _ESP32PIN<PIN, PIN> {};

_DEFPIN_ESP32(12);
...
```

focalintent, 2018-06-23:
Except things like the bitbang'd spi output _does_ write to the pins - so there do need to be FastPin definitions there.  WS28xx leds aren't the only ones out there : )

samguyer, 2018-06-24:
Oh, duh. All of the drivers for clocked chips need to access the pins. Ok, I'll fix it up

MarkusBec, 2018-07-02:
any progress with this?