FastLED/FastLED #618: FastLED changing my array value

jellewie, 2018-06-27:
**What stuff is for**
Retrieved[0] = The first character retrieved (~ = 126)
Retrieved[1] = The second character retrieved (~ or M)

**What I do**
Data send '{WAVE}M'
First PrintLine retrieved = 126 {WAVE}
FillSolid 
PrintLine the same value as above = 0?

Data send '{WAVE}{WAVE}' (This will not do the fill solid)
First PrintLine retrieved = 126 {WAVE}
PrintLine the same value as above = 126 {WAVE} (I can only include that the FillSolid is indeed the problem)

**What I expected**
That the value was left unchanged and both Prints would be the same

**Specs**
Arduino mega 2560
FastLED version 3.001.007
Arduino 1.8.6 (Hourly build 2018/05/28) but also confirmed on a older version

[image]

**The full Code**
```
#include "FastLED/FastLED.h"                                        //Include the FastLED library to control the LEDs in a simple fashion

const byte PWO_LED = 5;                                             //Where the <LED strip> is connected to
const byte PDO_Emergency = 53;                                      //Emergency button feedback
const int TotalLEDs = (48 + 36) * 2;                                //The total amount of LEDS in the strip

CRGB LEDs[TotalLEDs];                                               //This is an array of LEDs. One item for each LED in your strip.
byte Retrieved[3];                                                  //The array where we put the com data in
bool OverWrite;                                                     //If we need to overwrite the self thinking
bool LED_SensorDebug;                                               //If we need to debug the sensors (show sensor data in the LEDs)

void setup() {
  pinMode(PWO_LED, OUTPUT);
  Serial.begin(9600);                                               //Opens serial port (to pc), sets data rate to 9600 bps
  FastLED.addLeds<WS2812B, PWO_LED, GRB>(LEDs, TotalLEDs);          //Set the LED type and such
  Retrieved[0] = 126;                                               //Fake the emergency button from the PC, (just once on boot so when you connect the PC the PC takes this over)
}
void loop() {                                                       //Keep looping the next code
  if (Serial.available() > 0) {                                     //https://www.arduino.cc/en/Reference/ASCIIchart to see the asci chart to know what numbers are what
    Retrieved[0] = Serial.read();                                   //Get Emergency info (1 = it's fine, !1 WE ARE ON FIRE!)
    delay(1);                                                       //Some delay so we are sure we retrieved the data
    Retrieved[1] = Serial.read();                                   //Read next data
    delay(1);                                                       //Some delay so we are sure we retrieved the data
    Serial.println("Retrieved A=" + String(Retrieved[0]));
    if (Retrieved[1] == 'M') {                                      //If "77" retrieved (Motor)
      fill_solid(&(LEDs[167]), 2, CRGB(0, 255, 0));                 //^^
    }
    Serial.println("Retrieved B=" + String(Retrieved[0]));
  }
  delay(1000);
}
```

jellewie, 2018-06-27:
For your confirmation, I can't seem to get V3.1.8 working with this code

```
In file included from C:\Users\user\Documents\Arduino\temp\temp.ino:4:0:
sketch\FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.008
 #    pragma message "FastLED version 3.001.008"
                     ^
C:\Users\user\AppData\Local\Temp\cci6t7Ii.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.startup+0x1d6): undefined reference to `CLEDController::m_pHead'
<artificial>:(.text.startup+0x1da): undefined reference to `CLEDController::m_pHead'
<artificial>:(.text.startup+0x1e6): undefined reference to `CLEDController::m_pHead'
<artificial>:(.text.startup+0x1ea): undefined reference to `CLEDController::m_pHead'
<artificial>:(.text.startup+0x1ee): undefined reference to `CLEDController::m_pTail'
<artificial>:(.text.startup+0x1f2): undefined reference to `CLEDController::m_pTail'
<artificial>:(.text.startup+0x206): undefined reference to `CLEDController::m_pTail'
<artificial>:(.text.startup+0x20a): undefined reference to `CLEDController::m_pTail'
<artificial>:(.text.startup+0x236): undefined reference to `CFastLED::addLeds(CLEDController*, CRGB*, int, int)'
<artificial>:(.text.startup+0x2e2): undefined reference to `fill_solid(CRGB*, int, CRGB const&)'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

chrisparton1991, 2018-06-27:
This is an off-by-one error in your code. You have 168 LEDs. By doing a fill_solid on LED at index 167 for 2 LEDs, you're writing off the end of the array, which is changing your Retrieved array, which happens to be next to your LEDs array.

Remember that array indexes start at zero, so index 167 is the last element in the array.

jellewie, 2018-06-27:
Indeed, thans for the quick responce

This can be marked as fixed

chrisparton1991, 2018-06-27:
No problem, best of luck with the rest of your project! You should consider posting the results in the FastLED community page :)

closed 2018-07-06 by jellewie