FastLED/FastLED #619: FastLED.show() causes servo to twitch

tomchambers2, 2018-06-27:
When running FastLED.show() every 50 miliseconds (by measuring the delta rather than with a delay) and having a servo attached without the servo being sent any code, the servo twitches. This only happens when `.show()` is fired.

```
#include <Servo.h>
#include <FastLED.h>

#define LED_TYPE    WS2812B
#define LED_STRIP_PIN     6
#define NUM_LEDS    100
#define COLOR_ORDER GRB
#define LED_UPDATE_TIME 50

int lastLedUpdate = 0;

CRGB leds[NUM_LEDS];

Servo servo;

setup() {
  servo.attach(10);
  FastLED.addLeds<LED_TYPE, LED_STRIP_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.show();  
}

loop() {
  if (millis() - lastLedUpdate < LED_UPDATE_TIME) return;

  lastLedUpdate = millis();

  FastLED.clear();
  
  for (int i = 0; i <= NUM_LEDS; i++) {
    leds[i] = CRGB::Red;
  }

  FastLED.show();
}
```

focalintent, 2018-06-27:
Unfortunately, servos are driven by interrupts which get disabled while writing out ws2812 data.   Either switch to an led chipset that doesn’t require disabling interrupts like the apa102 or switch to a platform that allows interrupts to run while writing out led data like the teensy 3.x.

focalintent, 2018-06-27:
See https://github.com/FastLED/FastLED/wiki/Interrupt-problems for more info

focalintent, 2019-01-20:
https://learn.adafruit.com/neopixels-and-servos/the-ticoservo-library was mentioned as something that might work around this.

closed 2019-01-20 by focalintent