FastLED/FastLED #620: Servo/Interrupts Solution

J3698, 2018-06-30:
In the wiki under "Interrupt problems," I see that 

> If the lack of serviceable interrupts is causing problems for servos, you are out of luck - make one of the hardware switches mentioned above, or use a dual-controller setup where one avr talks to servos, and the other avr talks to the leds.

However Adafruit made a new servo library to fix this called TicoServo that works with a bunch (not all) of the Arduino boards, perhaps that should be mentioned.
https://learn.adafruit.com/neopixels-and-servos/the-ticoservo-library

focalintent, 2019-01-20:
Added to the wiki

closed 2019-01-20 by focalintent

damiannogueiras, 2021-01-05:
The library [PWMServo](https://platformio.org/lib/show/600/PWMServo) work well with FastLed.
Servo sporadic movements disappear.