FastLED/FastLED #622: APA102 strips with 288 LEDs

systembolaget, 2018-07-03:
What has to change in code/library regarding running effects along those? Even with highter NUM_LEDS defined, the effects don't "spill over" into the LEDs beyond 144.

Thanks for some hints as to what to do!

chrisparton1991, 2018-07-04:
Judging by your number of LEDs being 288, I'm inclined to think you're using 144 LED/m strips? If so, have you checked the connection between the strips? If you're not already, you could try feeding power into the second strip as well as the first.

This is all speculation though. As @marmilicious said, more info would be useful. I'd be interested to see a photo of your setup, particularly the points at which the strip(s) are being powered and how the strip(s) are being connected to the Arduino.

chrisparton1991, 2018-07-04:
Okay, so I've had a chance to look at the code, and the issue is that you're using 8-bit integers and `random8 ` with a `NUM_LEDS` value of 288.

The `random8` function takes a `uint8_t`, so calling `random8(NUM_LEDS)` is wrapping `NUM_LEDS` around to 32. This means that your `ledStart` value will always be between 0 and 31.

To fix, simply change your `ledstart` and `ledlen` to be `uint16_t` and assign them values using `random16` instead of `random8`.

Just to share how I approached this, I inserted this code after the first `FastLED.show()` call to see if anything was being populated in the leds past 144, and the fact they were all coming back as zero told me that the problem was with the code and not FastLED or the strips:
```
for (int i = 144; i < NUM_LEDS; i++) {
  Serial.print(leds[i].r + leds[i].g + leds[i].b);
  Serial.print(" ");
}
```

Let me know how you go, and once you've got it working you should consider sharing your progress on the [FastLED community](https://plus.google.com/communities/109127054924227823508) :)

systembolaget, 2018-07-04:
Chris, very much appreciated.

I was obviously not considering that an 8 bit integer can only hold values from 0 - 255. 288 - 255 = 33.

I better buy an Arduino C book for beginners ; )

closed 2018-07-04 by systembolaget