FastLED/FastLED #623: Error Message whilst uploading

JustrednaxG, 2018-07-04:
Hi,
when I try to upload your code I get this error message: 

exit status 1
'TaskHandle_t' does not name a type

Could you help me please?
Thanks in regards.

chrisparton1991, 2018-07-04:
Hi @JustrednaxG, what type of controller (e.g. Arduino Uno, ESP32) are you using? `TaskHandle_t` is used for ESP32 boards only, so it's possible you've copied some ESP32 code and tried to run it on an Arduino.

If this is your first time using FastLED, consider trying the [Blink demo](https://github.com/FastLED/FastLED/blob/master/examples/Blink/Blink.ino) as that will be easier to get set up, just follow the comments in the source to set up the correct parameters for your gear.

JustrednaxG, 2018-07-04:
Thank you, I will try this out.

JustrednaxG, 2018-07-04:
I am using an Arduino Uno for prototyping but I am planning to use an Arduino Nano from Elegoo

JustrednaxG, 2018-07-04:
@chrisparton1991 Thank you for your help, I now have better understanding of the FastLED library and have succesfully used different self-made and pre-made sketches. I will recommend your YouTube channel to anyone having problems with these leds who I know!
Thank you and the best of luck with your YT channel(and the staircase project)

chrisparton1991, 2018-07-04:
Awesome @JustrednaxG, thanks a lot! I'm working hard on Staircase V2, just got to find time around work and other commitments :)

If you want to see other cool FastLED stuff, be sure to check out the FastLED community at http://fastled.io/+. It's active and everyone's really friendly, you can show off your own stuff and ask questions there too!

closed 2019-01-20 by focalintent