FastLED/FastLED #626: ESP32 Clockless issues

tavrinnn, 2018-07-12:
Hardware: NodeMCU

My ESP32 crashes when attempting to invoke FastLED.Show(). The error message is as follows:
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

This is using the latest clockless_esp32.h on the master branch.

I dug through the previous issues and used an earlier version of clockless_esp32.h and that version doesn't seem to crash. The only problem with that version though is that when you use all 8 channels it randomly hangs when attempting to show().

Here is the link to that version: https://raw.githubusercontent.com/samguyer/FastLED/master/platforms/esp/32/clockless_esp32.h

Master811129, 2018-07-12:
Hardware: ESP32 Dev Module
IDE :1.8.5

i have this issue too please help.

Master811129, 2018-07-13:
ok i got it
add this code to the first line of your project (even the "#include <FastLED.h")
#define FASTLED_ALLOW_INTERRUPTS 0
thanks to @marcmerlin 
he solved this problem. but what does this code mean? does it disable the interrupts?

marcmerlin, 2018-07-13:
FASTLED_ALLOW_INTERRUPTS 0 
turns off interrupts while LEDs are being updated. You still get interrupts after the strip had been updated, but not during.
Why you need this on ESP32, no idea. ESP32 is very cool, but also complicated, and gives crashes that are not always what you expect (like the wrong version of FastLED causing crashes in totally unrelated library code)
You should talk to @samguyer since you are using his version of the code which should work on ESP32 (I have used it to display on a 32x32 matrix (neopixel yesterday, and it worked ok for me without disabling interrupts)

samguyer, 2018-07-13:
@marcmerlin @Master811129 @tavrinnn 

Hi all! You should not need to worry about disabling interrupts at all using my implementation of clockless_esp32.h, (which is now the default provided in the master branch of FastLED). In fact, I'm pretty sure that the FASTLED_ALLOW_INTERRUPTS directive is not used anywhere in the code.

Also: the latest version can support any number of strips -- it assigns them to the eight available RMT channels on-the-fly using a simple worklist algorithm.

One very useful utility for ESP32 problems is the ESP exception decoder. I highly recommend it:

https://github.com/me-no-dev/EspExceptionDecoder

Feel free to ping me if you are still having problems!

tavrinnn, 2018-07-14:
As @samguyer said above, disabling interupts won't achieve anything in the clockless_esp32 block as his implementation uses the RMT module to generate the precisely timed parallel led timings.

@samguyer what version of clockless_esp32 should I be using? the version on the master branch? I'll give the exception decoder a go and see what happens too

samguyer, 2018-07-14:
@tavrinnn The current version in the main FastLED branch is good, although I have some updates that should be merged in soon (waiting on a pull request!). If you want the latest-and-greatest you can use my branch.

tavrinnn, 2018-07-14:
Cheers, I'll give it a go and report back

tavrinnn, 2018-07-16:
Alright so I spent the whole afternoon playing with your latest master branch @samguyer 
I also added the ESP32 Exception handler toolset to the Arduino IDE.

Below is what I got when the ESP32 crashes:

tavrinnn, 2018-07-16:
I have managed to narrow the problem down a bit more

If WiFi.begin() isn't called, FastLED.show() will run flawlessly with the current clockless_esp32 implementation. However, the moment WIFI.begin() gets invoked, the chip will randomly crash when invoking FastLED.show().

Interestingly, the stack trace is only displayed when the FastLED.show() task is assigned to Core0. When I assign it to core 1 no stack trace is displayed - the chip simply just locks up entirely and becomes unresponsive.

I can provide a test sketch to replicate the problem if required.

samguyer, 2018-07-16:
@tavrinnn If you can send me a sketch that reproduces the problem, that would be great. It's surprising that you're seeing this problem even when FastLED runs on the other core. I'll check it out.

tavrinnn, 2018-07-16:
#include <FastLED.h>
#include <WiFi.h>

#define MAX_BRIGHTNESS        10
#define MAX_PIXEL_CHANNELS    8
#define MAX_PIXELS_IN_ROW     144
CRGB led_buff[MAX_PIXEL_CHANNELS * MAX_PIXELS_IN_ROW];

#define FAST_LED_REFRESH_CORE 0

static TaskHandle_t FastLEDRFTH;
static TaskHandle_t MainTH;
static int          FastLEDDelay;

unsigned long frameTime;
uint8_t gHue = 0;

WiFiUDP udpServer;

void simpleRefresh( int delayMS )
{
  MainTH = xTaskGetCurrentTaskHandle();
  FastLEDDelay = delayMS;
  xTaskNotifyGive(FastLEDRFTH);
  ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
}

void FastLEDRefreshFrame( void *pvParameters )
{
  unsigned long fStart, fEnd;

  while (1)
  {
    fStart = micros();
    ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
    FastLED.show();

    if (FastLEDDelay > 0)
      FastLED.delay(FastLEDDelay);

    xTaskNotifyGive(MainTH);
    fEnd = micros();
    frameTime = fEnd - fStart;
  }
}

void setup() {
  Serial.begin(921600);

  FastLED.addLeds<WS2812B, 13, GRB>(led_buff, 0, MAX_PIXELS_IN_ROW);
  FastLED.addLeds<WS2812B, 12, GRB>(led_buff, 1 * MAX_PIXELS_IN_ROW, MAX_PIXELS_IN_ROW);
  FastLED.addLeds<WS2812B, 14, GRB>(led_buff, 2 * MAX_PIXELS_IN_ROW, MAX_PIXELS_IN_ROW);
  FastLED.addLeds<WS2812B, 27, GRB>(led_buff, 3 * MAX_PIXELS_IN_ROW, MAX_PIXELS_IN_ROW);
  FastLED.addLeds<WS2812B, 26, GRB>(led_buff, 4 * MAX_PIXELS_IN_ROW, MAX_PIXELS_IN_ROW);
  FastLED.addLeds<WS2812B, 25, GRB>(led_buff, 5 * MAX_PIXELS_IN_ROW, MAX_PIXELS_IN_ROW);
  FastLED.addLeds<WS2812B, 19, GRB>(led_buff, 6 * MAX_PIXELS_IN_ROW, MAX_PIXELS_IN_ROW);
  FastLED.addLeds<WS2812B, 22, GRB>(led_buff, 7 * MAX_PIXELS_IN_ROW, MAX_PIXELS_IN_ROW);
  FastLED.setBrightness(MAX_BRIGHTNESS);
  xTaskCreatePinnedToCore(FastLEDRefreshFrame, "FastLEDRefresh", 2048, NULL, 2, &FastLEDRFTH, FAST_LED_REFRESH_CORE);

  // WiFi init causes crashes on FastLED.show() (Line 411 in Clockless_ESP32.h)
  WiFi.begin("ESP32", "12345678");
  WiFi.setSleep(false);
  udpServer.begin(2302);
}

void rainbow() {
  for (int i = 0; i < MAX_PIXEL_CHANNELS; i++) { fill_rainbow(led_buff + (i * MAX_PIXELS_IN_ROW), MAX_PIXELS_IN_ROW, gHue, 7); }
}

void loop() {
    rainbow();
    simpleRefresh(1);
    EVERY_N_MILLISECONDS(10) { gHue++; }
}

samguyer, 2018-07-16:
Ok, I'll try it out today and let you know what I find.

tavrinnn, 2018-07-16:
No worries. Thanks :D

samguyer, 2018-07-17:
@tavrinnn I tried your code, and I was able to reproduce the crash you described. Interestingly, it only happens when FastLED.show() is running on the *other* core (Core 0), while the main thread and the WiFi are running on Core 1. When I run everything on Core 1, it seems to work fine. That makes me think it has more to do with synchronization than interrupts.

So, I tried starting with the DemoReelESP32 example that I added to the examples directory. I made sure it spawned the show task on Core 0, and added the WiFi initialization, and it seems to work. Maybe grab the task functions from that code and paste them into your code.

Let me know how it goes.

tavrinnn, 2018-07-17:
Sure I'll give that one a go and report back later today

tavrinnn, 2018-07-17:
After a bit of fiddling I decided to ditch the xTaskCreatePinnedToCore() implementation and simply invoked FastLED.show() within the context of the original loop task. It looks pretty promising so far.

I could get the DemoReelESP32 sketch to crash when calling the FastLEDshowESP32() function as well... although with the FastLED.delay() set to a sensible limit in the DemoReel sketch it takes longer for the problem to appear.

I'll leave it running overnight and see if it crashes. Thanks @samguyer

samguyer, 2018-07-17:
@tavrinnn I will continue to investigate what's going on in your original code. There is definitely something going on there, although it is strange that it only happens when FastLED.show() is running on the *other* core.

tavrinnn, 2018-07-17:
Ah... the success was short lived. It appears that when its invoked directly within loop() the chip just hangs randomly and becomes unresponsive

samguyer, 2018-07-19:
@tavrinnn Can you pull the latest checkin from my repo? The git commit message is "Experimental". Let me know if it changes the behavior at all. Thanks!

tavrinnn, 2018-07-19:
Giving it a go now - Ill let you know how it goes

tavrinnn, 2018-07-19:
Well... there's good news and bad news... Good news is that it runs perfect. No locking up on Core1 and no exceptions being thrown on Core0. Onto the Bad news... and that's where it ends. There is none :D

I'll do more testing tomorrow and over the weekend and report back if I come across any strange errors but so far it looks and works great! Cheers!!!!!

samguyer, 2018-07-19:
@tavrinnn Ha! You totally got me. Glad to hear it. I'll have to look more closely at why my change fixes the problem. I think it's a memory management and/or memory corruption problem.

tavrinnn, 2018-07-22:
The ESP32 has been running flawlessly over the weekend. Happy to close this ticket @samguyer if you're happy with your changes.

Cheers!

Bedmonds91, 2018-10-18:
Hello, I have the exact same issue as tavrinnn while using WiFi and ESPAsyncWebServer. I updated the FastLED but I still get the same Guru Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0) message. The code is essentially the same with a FastLEDshowESP32 function call to run FastLEDshow() on Core 0. I don't have any delays around FastLEDshowESP32 in the main loop. Is there something I'm missing. Let me know if you need my code.

samguyer, 2018-10-18:
@Bedmonds91 I'm happy to take a look. One simple thing you can test is just comment out the FastLED.show() in the show task (the function running on the other core).

Bedmonds91, 2018-10-18:
Ok, it works flawlessly without FastLED.show() and I get the panic error when I use the show function on either core. The only difference is when the show is on Core 0, there is something causing a 200ms delay with the wifi event handler. I'll also add that I'm using an app with a scroller to send values over softAP instantaneously. When sending the continuous values with the scroller, the program just crashes sooner than when leaving it alone. I tried adding FastLED.delay(5) after the show command in both core 1 and core 0 and get those two different results. 
[ESP32_FastLED_AsyncServer.zip](https://github.com/FastLED/FastLED/files/2493104/ESP32_FastLED_AsyncServer.zip)

Bedmonds91, 2018-10-18:
The Core 0 task function is in utils.h and the WiFi event handler is in WifiStuff.h.

samguyer, 2018-10-18:
@Bedmonds91 Just so I understand the architecture, what code is running on each core?

Bedmonds91, 2018-10-18:
Sorry, with the show function on core 0, there is just the crash at the start of the program then after a few seconds just a 200ms delay. I'm measuring this by setting a wifievent flag and displaying heap size, the changing value, and the time delay (which is exactly 200ms). Below is the register dump I got at the beginning of the program.

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40081d38  PS      : 0x00050034  A0      : 0x400d6261  A1      : 0x3ffbe1c0  
A2      : 0x00002048  A3      : 0x0001a049  A4      : 0x00002048  A5      : 0x3ffbe190  
A6      : 0x00000008  A7      : 0x3ff56000  A8      : 0x00000000  A9      : 0x3ffc1df8  
A10     : 0x00000001  A11     : 0x00000001  A12     : 0x0000001f  A13     : 0x00000100  
A14     : 0x00000000  A15     : 0x3ffcc840  SAR     : 0x00000019  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400d6122  LEND    : 0x400d6144  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x400d6261  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40081d38

Backtrace: 0x40081d38:0x3ffbe1c0 0x400d625e:0x00000000

Core 1 register dump:
PC      : 0x400e7986  PS      : 0x00060334  A0      : 0x8008e310  A1      : 0x3ffbc670  
A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x80000020  
A6      : 0x00000000  A7      : 0x3ffbd650  A8      : 0x3ffc31a8  A9      : 0x3ffc316c  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x8008c658  A13     : 0x3ffbc570  
A14     : 0x00000000  A15     : 0x3ffbc340  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400e7986:0x3ffbc670 0x4008e30d:0x3ffbc690 0x4008c445:0x3ffbc6b0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1496
load:0x40078000,len:8596
load:0x40080400,len:6980
entry 0x400806f4

Bedmonds91, 2018-10-18:
Ok so I just realized the 200ms delay is coming from the ulTaskNotifyTake(pdTRUE, 200). So FastLED.show is taking longer than 200ms to complete. After changing it to ulTaskNotifyTake(pdTRUE, portMAX_DELAY) the program now crashes if I send the server too many values at once. I also have been experimenting with FastLED.delay() in core 0 and it doesn't help.

Bedmonds91, 2018-10-18:
So the rest of the program works fine if I set the blocking time to something short like 40 -> ulTaskNotifyTake(pdTRUE, 40). So something with the WiFi server is making the FastLED.show get stuck in a loop somewhere.

Bedmonds91, 2018-10-18:
Can confirm that FastLED.show freezes after about a minute while connected to my phone when acting as a soft AP.

samguyer, 2018-10-19:
OK, so here's another thing you can try: in FastLEDshowESP32() don't wait for the notification at all. This will cause FastLED.show() to run in parallel with the main thread. Since you're not waiting for it to finish, you just need to be careful not to call it again too soon.

X-WL, 2018-10-19:
Hello! I experience this problem when using Wifi and ESPAsyncWebServer. However, this simply leads to a hang without a kernel panic. Previously, when I did not use ESPAsyncWebServer, the Execution Code on the other core caused a hang. but it was possible to reduce them simply by calling FastLED.show () from the main loop.

Bedmonds91, 2018-10-19:
Ok so the problem still occurs when I take out the notification. That just allows the other loop to run unimpeded. I tried multiple combinations of delay and FastLED.delay in both the core 1 and core 0. Should I maybe try putting the show function in the main loop with a yield()?

samguyer, 2018-10-19:
Give that a try. I looked over the ESPAyncWebServer code, and I did notice a couple of things about the underlying AsyncTCP library. For one, it runs on Core 1, which is a little strange because the ESP32 IDF core uses Core 1 as the default for the main loop. So, if you're using AsyncTCP through the Arduino interface, you're not actually using the second core (Core 0) at all. But that doesn't explain why there is a conflict between FastLED and AsyncTCP. The only other thing I see is that the AsyncTCP task has a higher priority than the FastLED show task. So, you could bump the 2 to a 3 in xTaskCreatePinnedToCore.

Bedmonds91, 2018-10-19:
Ok so changing Core 0 privilege to 3 didn't help, but it works much better when I have show() at the end of the main loop and surrounded by yield(). When using the scroller in the app, I found that the wdt will panic and reset the chip if I send the values too fast which can sometimes cause the wifi to switch to a different connection. Is there a way to disable the watchdog timer? I tried ESP.wdtFeed() and esp_task_wdt_reset() and I get compilation errors saying I don't have access to those functions or that they don't exist.

Bedmonds91, 2018-10-19:
I did some more testing and it will panic even when I send values at a slower rate. And as long as my phone doesn't automatically switch to a different AP the program continues to receive commands like normal. Another thing is that I haven't actually tested to see if the show function is updating the actual pixels but I can check tonight when I'm home. I'm currently testing on an ESP32 dev board and seeing checking the serial monitor. The show function was always hanging up when on core 0 and now its just making the wdt fire randomly but only when I send values. If I leave it alone it seems to be fine. I think the show function crashes when the wifi handler receives a value while the program is in the show function.

samguyer, 2018-10-19:
That is really weird. How hard would it be for me to replicate your setup myself?
Also: do you have the ESP32 Backtrace decoder, so you can see where the panic happens?

Bedmonds91, 2018-10-19:
You could modify my code to include a web browser similar to the WiFiAccessPoint sketch from ESP core WiFi library examples. Then just continuously press the command buttons until it crashes. I'll see if I can make those modifications myself and send you the updated sketch. I'll also look into the Backtrace encoder and see what I get. I'm currently using thunkable app to push the url commands to the server so I could share that with you if you want.

Bedmonds91, 2018-10-19:
So everything works pretty well right now as long as I limit the server requests from the app. Is there a way to block the FastLED.show() function or maybe the AsyncWebServer handler ?

tschundler, 2019-11-24:
Not sure if this is a good place to post this, but I ran into a similar issue myself recent - isr wdt timeout on FastLED's show. The problem seemed to be related to the interplay between the producer task and the ISR. Running FastLED.delay() or .show() from a task with a high priority solved the problem, e.g.

```
  xTaskCreatePinnedToCore(&leds_task, "leds_task", 4096, NULL, 100, NULL, 1);
```

That somehow saved the ISR from getting starved while another process stole the core with the LEDs for a bit handling http/whatever.