FastLED/FastLED #627: Arduino Zero Not working

interrgned, 2018-07-14:
Arduino Zero Using this library does not working
LED:WS2812B

chrisparton1991, 2018-07-14:
Hi @interrgned, what version of the arduino IDE are you using? And have you had success using FastLED on a different Arduino?

interrgned, 2018-07-14:
oh
Arduino IDE version 1.8.5
Library version 3.18
using ESP8266 working

interrgned, 2018-07-15:
Measuring the output signal with an oscill
No signal at all

(Use Adafruit NeoPixel Library
The output is only ~760KHz
Same as not working)

chrisparton1991, 2018-07-15:
Interesting. According to #254 the Zero is supported. I don't have one to test with, unfortunately.

Have you tried using different I/O pins? According to #327, pins 4 and 5 didn't work for the poster, so perhaps only some pins will work correctly.

interrgned, 2018-07-15:
Tested pins 0 2 3 and 12
Can't work……

interrgned, 2018-07-15:
https://github.com/FastLED/FastLED/issues/546
I tried this modification and it didn't work

chrisparton1991, 2018-07-15:
Bummer. Can you share the sketch you're using?

interrgned, 2018-07-15:
“Example DemoReel100”
`#include "FastLED.h"

FASTLED_USING_NAMESPACE

// FastLED "100-lines-of-code" demo reel, showing just a few 
// of the kinds of animation patterns you can quickly and easily 
// compose using FastLED.  
//
// This example also shows one easy way to define multiple 
// animations patterns and have them automatically rotate.
//
// -Mark Kriegsman, December 2014

#if defined(FASTLED_VERSION) && (FASTLED_VERSION < 3001000)
#warning "Requires FastLED 3.1 or later; check github for latest code."
#endif

#define DATA_PIN    3
//#define CLK_PIN   4
#define LED_TYPE    WS2812B
#define COLOR_ORDER GRB
#define NUM_LEDS    100
CRGB leds[NUM_LEDS];`

chrisparton1991, 2018-07-15:
If you're using the standard demoreel you probably don't have any issues with the sketch.

Unfortunately I have to head to bed now, got to work tomorrow.

The only advice I can give you for now is to ensure you're using the latest FastLED version, and perhaps try to get a simpler sketch like the Blink example working, to help rule out any code issues.

Failing that, you'll have to wait for one of the team members or someone with Arduino Zero experience.

You might get help quicker if you try posting your question under Support in the FastLED community page at http://fastled.io/+.

Best of luck!

interrgned, 2018-07-15:
Thank you very much
good night

interrgned, 2018-07-17:
Zero output signal:
Data 1 ~ 660 KHz 1.500us
Data 0 ~ 640 KHz 1.572us
Frequency will change
And the wavelength is over 1.25us plus or minus 150ns
Not working
The output signal of the ESP8266 is ~730 KHz period 1.372uS
Frequency is stable
Can work

focalintent, 2018-07-17:
I wonder if the instruction timings are different on the D21 M0+ than other M0+ cores I've worked on? One possible thing to try would be to go into the library headers and remove this line:

https://github.com/FastLED/FastLED/blob/1b44dfa547035e50e3706e0f43c73fa6c460fb17/platforms/arm/d21/led_sysdefs_arm_d21.h#L6

```
#define FASTLED_ARM_M0_PLUS
```

(If you're curious at a low level what that gates - on some M0's a branch instruction is 3 cycles, but on the M0+ it's only 2 - so there's some padding work that gets done on the M0+ to keep the timings in line with the M0 instruction timings as far as when bits get flipped.   Of course, if, in spite of being labeled as an M0+, the Arduino Zero is actually an M0 - then those extra padding opcodes will slow the output timing by adding a few clock cycles to each bit's output - which isn't what you want here : ) 

At least, without a zero here that I can run and test with and look at the compiler output on, I'm speculating. If that doesn't fix the issue, then there may be other instruction timing issues that are at play that make the zero different from other M0's.

interrgned, 2018-07-17:
Now see the output signal with an oscilloscope
The signal is normal
But the LED still doesn't work...
strange

interrgned, 2018-07-17:
Measured signal
Data0 HIGH 392ns LOW 932ns
Data1 HIGH 996ns LOW 312ns
is correct
LED still does not work……

focalintent, 2018-07-17:
I'm not sure why your leds still aren't working - it's possible you may need a 3.3v to 5v level shifter.  As for the difference in timing, I think the cause may actually be the D21 having single cycle port i/o register write times (the M0 asm code assumes 2 cycle write times - the change I had you make above happens to shift the total timing count in a way that accounts for this, which is why it works, but that should get fixed at some point in the future).  Also - you have a common ground between the zero and your leds?

interrgned, 2018-07-18:
[image]
Can work!
Use 74HC245 bus buffer
Thank you for your help.

closed 2018-09-08 by focalintent