FastLED/FastLED #634: Indexing of WS2813 pixel strip is off

alexdmiller, 2018-08-03:
I'm experiencing a strange problem when indexing pixels on my WS2813 pixel strip. The indices work fine up until index 63. At that point, they start jumping around. Here's what I mean:

The following code should turn pixel at index 63 to blue:

```
#include <FastLED.h>

#define LED_PIN     5
#define NUM_LEDS    74

#define BRIGHTNESS  255

#define LED_TYPE    WS2813
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];

#define UPDATES_PER_SECOND 100

void setup() {
    delay( 3000 ); // power-up safety delay
    FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
    FastLED.setBrightness(  BRIGHTNESS );
}

void loop()
{
  for (int i = 0; i < NUM_LEDS; i++) {
    leds[i] = CRGB::Black;
  }

  leds[63] = CRGB::Blue;
  
  FastLED.show();
  FastLED.delay(1000 / UPDATES_PER_SECOND);
}
```

Instead, it actually skips a couple of pixels, and turns the pixel at index 65 blue! Even more strangely, setting `leds[64] = CRGB::Blue;` does what is expected, but then `leds[65]` skips again. This continues for the remaining length of the pixel strip.

I can get the indices to behave normally If I turn all the pixels white, like so:

```
for (int i = 0; i < NUM_LEDS; i++) {
  leds[i] = CRGB::White;
}

leds[63] = CRGB::Blue;
```

I don't know enough about FastLED/WS2813 to diagnose the problem, but any help would be greatly appreciated.

marmilicious, 2018-08-04:
Is this happening on single continuous strip, or do you have a 60 pixel strip and then another connected to that?

alexdmiller, 2018-08-04:
It was with multiple shorter strips connected together. I think this was caused by some back up data pins not being connected — we re-soldered everything and it works!

closed 2018-08-04 by alexdmiller