FastLED/FastLED #638: Strange flicker on fade out with neopixels

rorosaurus, 2018-08-11:
I see this strange behavior during animations which fade to black on the my neopixels.

I know neopixels have some strange behavior at low brightness settings, do we know if that is the problem?

[image]

Code is the DemoReel100 FastLED example.  .gif above is slowed 50% to make it easier to see.

rorosaurus, 2018-08-15:
I was able to confirm a similar flicker on my WS2812B strip as well during this example demo, so I no longer think it is related to the LED type.

DrBit, 2018-08-15:
I can also confirm a similar behaviour in one of my simple examples. Maybe in my case is due to using an ESP32? It seemed that it started when i used wifi together with  the ws2812. I have to further test it

DrDiodac, 2018-08-15:
I also confirm.. horrible flicking/blinking witch WS2812B, i tested also other settings without success. 
My code runs on Teensy 3.1. Any suggestion/tips how to overcome problems like that??

rorosaurus, 2018-08-16:
Interestingly, I wasn't able to reproduce this issue on my Trinket-powered neopixel goggles...

rorosaurus, 2018-08-17:
Reproduces whether I power the LEDs from 5V or 3.3V, so not a data signal strength ratio issue I think.

rorosaurus, 2018-08-22:
Hi everyone, I found a solution from the helpful google+ FastLED community!

Change your include for FastLED to:
````
#define FASTLED_ALLOW_INTERRUPTS 0
#include "FastLED.h"
````

It seems like this issue is called by interrupts being called while fastLED is bitbanging the output to the strips!

DrDiodac, 2018-08-22:
If I reproduce my code on ArduinoMega all works very well but on Teensy 3.2 not really, blinking unreasonable but from time when I put my comment about problems I did some progress with bit shifting and timing. Is better, leds now stop blinking just little bit dimming   xD
I not give up and still digging for what I want just smooth changes colours and sequences of columns and rows according to control  from touch screen. If I get proper working for Teensy I let you guys know how to do with Teensy boards.

DrDiodac, 2018-08-22:
#define FASTLED_ALLOW_INTERRUPTS 0
#include "FastLED.h"

In my case, that solution change nothing:-/

focalintent, 2018-08-22:
Also try turning off dithering - it’s supposed to turn off if the refresh rate is below 100hz, but that logic may be faulty.

DrDiodac, 2018-08-25:
I downloaded from PJRC WS2811Serial lib and go by basic configuration  and now all working perfect without any problems and additional code lines for dithering or interrupts. 
WS2811Serial + FastLED is really the best option for Teensy family boards. And if all setup is done correctly, all working nice without any additional work around clock setting. Thank you guys for supporting.

focalintent, 2018-09-08:
FastLED's allowing of interrupts is intentional - and there's a mechanism to turn it off if it is interfering with your setup.

closed 2018-09-08 by focalintent