FastLED/FastLED #639: Support for Feather M4 Express (SAMD51)

pickledgator, 2018-08-15:
Hi all, is there listed support for the newer Feather M4 Express boards from Adafruit? It seems that the M0 Express boards compile FastLED just fine, however, the SAMD51 boards throw a spew of conflicts.

```
Arduino: 1.8.5 (Mac OS X), Board: "Adafruit Feather M4 (SAMD51), Enabled"

Build options changed, rebuilding all
In file included from /Users/nicfischer/Documents/Arduino/blink/blink.ino:1:0:
/Users/nicfischer/Documents/Arduino/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.007
 #    pragma message "FastLED version 3.001.007"
                     ^
In file included from /Users/nicfischer/Documents/Arduino/libraries/FastLED/led_sysdefs.h:32:0,
                 from /Users/nicfischer/Documents/Arduino/libraries/FastLED/FastLED.h:44,
                 from /Users/nicfischer/Documents/Arduino/blink/blink.ino:1:
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/led_sysdefs_avr.h:16:32: error: conflicting declaration 'typedef volatile uint8_t RoReg'
 typedef volatile       uint8_t RoReg; /**< Read only 8-bit register (volatile const unsigned int) */
                                ^
In file included from /Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51.h:44:0,
                 from /Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd.h:143,
                 from /Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/sam.h:540,
                 from /Users/nicfischer/Library/Arduino15/packages/adafruit/hardware/samd/1.2.1/cores/arduino/Arduino.h:48,
                 from sketch/blink.ino.cpp:1:
/Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51j19a.h:54:33: error: 'RoReg' has a previous declaration as 'typedef volatile uint32_t RoReg'
 typedef volatile       uint32_t RoReg;   /**< Read only 32-bit register (volatile const unsigned int) */
                                 ^
In file included from /Users/nicfischer/Documents/Arduino/libraries/FastLED/led_sysdefs.h:32:0,
                 from /Users/nicfischer/Documents/Arduino/libraries/FastLED/FastLED.h:44,
                 from /Users/nicfischer/Documents/Arduino/blink/blink.ino:1:
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/led_sysdefs_avr.h:17:32: error: conflicting declaration 'typedef volatile uint8_t RwReg'
 typedef volatile       uint8_t RwReg; /**< Read-Write 8-bit register (volatile unsigned int) */
                                ^
In file included from /Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51.h:44:0,
                 from /Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd.h:143,
                 from /Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/sam.h:540,
                 from /Users/nicfischer/Library/Arduino15/packages/adafruit/hardware/samd/1.2.1/cores/arduino/Arduino.h:48,
                 from sketch/blink.ino.cpp:1:
/Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51j19a.h:61:33: error: 'RwReg' has a previous declaration as 'typedef volatile uint32_t RwReg'
 typedef volatile       uint32_t RwReg;   /**< Read-Write 32-bit register (volatile unsigned int) */
                                 ^
In file included from /Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/fastled_avr.h:6:0,
                 from /Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms.h:32,
                 from /Users/nicfischer/Documents/Arduino/libraries/FastLED/FastLED.h:55,
                 from /Users/nicfischer/Documents/Arduino/blink/blink.ino:1:
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h: In member function 'virtual void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showPixels(PixelController<RGB_ORDER>&)':
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:112:7: error: there are no arguments to 'cli' that depend on a template parameter, so a declaration of 'cli' must be available [-fpermissive]
   cli();
       ^
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:112:7: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:165:7: error: there are no arguments to 'sei' that depend on a template parameter, so a declaration of 'sei' must be available [-fpermissive]
   sei();
       ^
In file included from /Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51.h:44:0,
                 from /Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd.h:143,
                 from /Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/sam.h:540,
                 from /Users/nicfischer/Library/Arduino15/packages/adafruit/hardware/samd/1.2.1/cores/arduino/Arduino.h:48,
                 from sketch/blink.ino.cpp:1:
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h: In static member function 'static void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showRGBInternal(PixelController<RGB_ORDER>&)':
/Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51j19a.h:935:27: error: expected identifier before '(' token
 #define PORT              ((Port     *)0x41008000UL) /**< \brief (PORT) APB Base Address */
                           ^
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:192:6: note: in expansion of macro 'PORT'
     [PORT] "M" (FastPin<DATA_PIN>::port()-0x20),  \
      ^
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:210:101: note: in expansion of macro 'ASM_VARS'
 #define LDSCL4(B,O) asm __volatile__("ldd %[scale_base], Z + %[" #O "]\n\tclr %[" #B "]\n\tclc\n\t" ASM_VARS );
                                                                                                     ^
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:368:4: note: in expansion of macro 'LDSCL4'
    LDSCL4(b0,O0)  PRESCALEA2(d0)
    ^
/Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51j19a.h:935:27: error: expected ']' before '(' token
 #define PORT              ((Port     *)0x41008000UL) /**< \brief (PORT) APB Base Address */
[... 252 lines cut ...]
                   ^
/Users/nicfischer/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51j19a.h:935:27: error: expected string-literal before '(' token
 #define PORT              ((Port     *)0x41008000UL) /**< \brief (PORT) APB Base Address */
                           ^
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:192:6: note: in expansion of macro 'PORT'
     [PORT] "M" (FastPin<DATA_PIN>::port()-0x20),  \
      ^
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:236:54: note: in expansion of macro 'ASM_VARS'
 #define SCALE02(B,N) asm __volatile__( _SCALE02(B,N) ASM_VARS );
                                                      ^
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:369:19: note: in expansion of macro 'SCALE02'
    PRESCALEB4(d0) SCALE02(b0,0)
                   ^
In file included from /Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/fastled_avr.h:6:0,
                 from /Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms.h:32,
                 from /Users/nicfischer/Documents/Arduino/libraries/FastLED/FastLED.h:55,
                 from /Users/nicfischer/Documents/Arduino/blink/blink.ino:1:
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:192:10: error: expected ')' before ']' token
     [PORT] "M" (FastPin<DATA_PIN>::port()-0x20),  \
          ^
/Users/nicfischer/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:236:54: note: in expansion of macro 'ASM_VARS'
 #define SCALE02(B,N) asm __volatile__( _SCALE02(B,N) ASM_VARS );
                                                      ^
... (truncated for post length)
```

focalintent, 2018-08-18:
To borrow my comment from #640:

If the SAMD51 is effectively the same as the SAMD21 then there just needs to be a platform/pin definitions for the ItsyBitsy M4 added to https://github.com/FastLED/FastLED/blob/master/platforms.h and https://github.com/FastLED/FastLED/blob/master/led_sysdefs.h and https://github.com/FastLED/FastLED/blob/master/platforms/arm/sam/fastpin_arm_sam.h -- if the SAMD51 is significantly different in any way, then the library will be a bit more porting work (as loosely described in https://github.com/FastLED/FastLED/blob/master/PORTING.md but may be able to use the samd21 code as a reference).

focalintent, 2019-01-20:
SAMD51 support was merged in, so this should help

closed 2019-01-20 by focalintent

manitou48, 2019-04-14:
M4 express testing discussed at
https://github.com/FastLED/FastLED/issues/693

firepixie, 2019-06-23:
I'm trying to use the Adafruit Metro M4 AirLift Lite and having no luck.. I've got the latest version of FastLED and have tried making the changes above but am still getting errors.  Any help greatly appreciated!

```
WARNING: Spurious .tests folder in 'Adafruit IO Arduino' library
In file included from /var/folders/9k/jgng_3r53yx7sly3mn4wyccw0000gr/T/arduino_modified_sketch_349518/Blink.ino:1:0:
/Volumes/Home/Users/erin/Documents/Arduino/libraries/FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.008
 #    pragma message "FastLED version 3.002.008"
                     ^
In file included from /Volumes/Home/Users/erin/Documents/Arduino/libraries/FastLED/FastLED.h:65:0,
                 from /var/folders/9k/jgng_3r53yx7sly3mn4wyccw0000gr/T/arduino_modified_sketch_349518/Blink.ino:1:
/Volumes/Home/Users/erin/Documents/Arduino/libraries/FastLED/fastspi.h:115:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from /Volumes/Home/Users/erin/Documents/Arduino/libraries/FastLED/FastLED.h:48:0,
                 from /var/folders/9k/jgng_3r53yx7sly3mn4wyccw0000gr/T/arduino_modified_sketch_349518/Blink.ino:1:
/Volumes/Home/Users/erin/Documents/Arduino/libraries/FastLED/fastpin.h: In instantiation of 'class FastPin<3u>':
/Volumes/Home/Users/erin/Documents/Arduino/libraries/FastLED/platforms/arm/d51/clockless_arm_d51.h:19:49:   required from 'class ClocklessController<3, 279, 699, 419, (EOrder)10u, 0, false, 50>'
/Volumes/Home/Users/erin/Documents/Arduino/libraries/FastLED/chipsets.h:571:7:   required from 'class WS2812Controller800Khz<3u, (EOrder)10u>'
/Volumes/Home/Users/erin/Documents/Arduino/libraries/FastLED/FastLED.h:105:52:   required from 'class WS2812B<3u, (EOrder)10u>'
/Volumes/Home/Users/erin/Documents/Arduino/libraries/FastLED/FastLED.h:302:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2812B; unsigned char DATA_PIN = 3u; EOrder RGB_ORDER = (EOrder)10u]'
/var/folders/9k/jgng_3r53yx7sly3mn4wyccw0000gr/T/arduino_modified_sketch_349518/Blink.ino:22:61:   required from here
/Volumes/Home/Users/erin/Documents/Arduino/libraries/FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^
exit status 1
Error compiling for board Adafruit Metro M4 (SAMD51).
```

focalintent, 2019-06-24:
It looks like there aren’t pin definitions for the metro yet - also the m4 express doesn’t have a pin 3 defined it looks like (which definitely explains the error you are seeing if the metro is reading as an express)

firepixie, 2019-06-24:
There's a pin 3 labeled here on the Metro... ? I also tried a ton of other pin numbers and nothing seemed to work. What can I do to get pin definitions included?

firepixie, 2019-07-02:
OK I have added the pin definitions and created a pull request here: 

https://github.com/FastLED/FastLED/pull/834