FastLED/FastLED #640: Support for ItsyBitsy M4 (SAMD51)

vccabral, 2018-08-17:
I want to use FastLED with new ItsyBitsy. The board can only use pin 5 at 5v unless it is level shifted, not sure if that is relevant. 

This seems to be very similar to #639 

```
In file included from /var/folders/ns/936wknzn38x4vjh0l3z0sj740000gn/T/arduino_modified_sketch_375437/Blink.ino:1:0:
/Users/bracero/Documents/Arduino/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.007
 #    pragma message "FastLED version 3.001.007"
                     ^
In file included from /Users/bracero/Documents/Arduino/libraries/FastLED/led_sysdefs.h:32:0,
                 from /Users/bracero/Documents/Arduino/libraries/FastLED/FastLED.h:44,
                 from /var/folders/ns/936wknzn38x4vjh0l3z0sj740000gn/T/arduino_modified_sketch_375437/Blink.ino:1:
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/led_sysdefs_avr.h:16:32: error: conflicting declaration 'typedef volatile uint8_t RoReg'
 typedef volatile       uint8_t RoReg; /**< Read only 8-bit register (volatile const unsigned int) */
                                ^
In file included from /Users/bracero/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51.h:40:0,
                 from /Users/bracero/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd.h:141,
                 from /Users/bracero/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/sam.h:540,
                 from /Users/bracero/Library/Arduino15/packages/adafruit/hardware/samd/1.2.1/cores/arduino/Arduino.h:48,
                 from sketch/Blink.ino.cpp:1:
/Users/bracero/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51g19a.h:54:33: error: 'RoReg' has a previous declaration as 'typedef volatile uint32_t RoReg'
 typedef volatile       uint32_t RoReg;   /**< Read only 32-bit register (volatile const unsigned int) */
                                 ^
In file included from /Users/bracero/Documents/Arduino/libraries/FastLED/led_sysdefs.h:32:0,
                 from /Users/bracero/Documents/Arduino/libraries/FastLED/FastLED.h:44,
                 from /var/folders/ns/936wknzn38x4vjh0l3z0sj740000gn/T/arduino_modified_sketch_375437/Blink.ino:1:
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/led_sysdefs_avr.h:17:32: error: conflicting declaration 'typedef volatile uint8_t RwReg'
 typedef volatile       uint8_t RwReg; /**< Read-Write 8-bit register (volatile unsigned int) */
                                ^
In file included from /Users/bracero/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51.h:40:0,
                 from /Users/bracero/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd.h:141,
                 from /Users/bracero/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/sam.h:540,
                 from /Users/bracero/Library/Arduino15/packages/adafruit/hardware/samd/1.2.1/cores/arduino/Arduino.h:48,
                 from sketch/Blink.ino.cpp:1:
/Users/bracero/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51g19a.h:61:33: error: 'RwReg' has a previous declaration as 'typedef volatile uint32_t RwReg'
 typedef volatile       uint32_t RwReg;   /**< Read-Write 32-bit register (volatile unsigned int) */
                                 ^
In file included from /Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/fastled_avr.h:6:0,
                 from /Users/bracero/Documents/Arduino/libraries/FastLED/platforms.h:32,
                 from /Users/bracero/Documents/Arduino/libraries/FastLED/FastLED.h:55,
                 from /var/folders/ns/936wknzn38x4vjh0l3z0sj740000gn/T/arduino_modified_sketch_375437/Blink.ino:1:
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h: In member function 'virtual void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showPixels(PixelController<RGB_ORDER>&)':
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:112:7: error: there are no arguments to 'cli' that depend on a template parameter, so a declaration of 'cli' must be available [-fpermissive]
   cli();
       ^
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:112:7: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:165:7: error: there are no arguments to 'sei' that depend on a template parameter, so a declaration of 'sei' must be available [-fpermissive]
   sei();
       ^
In file included from /Users/bracero/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51.h:40:0,
                 from /Users/bracero/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd.h:141,
                 from /Users/bracero/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/sam.h:540,
                 from /Users/bracero/Library/Arduino15/packages/adafruit/hardware/samd/1.2.1/cores/arduino/Arduino.h:48,
                 from sketch/Blink.ino.cpp:1:
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h: In static member function 'static void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showRGBInternal(PixelController<RGB_ORDER>&)':
/Users/bracero/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/samd51g19a.h:897:27: error: expected identifier before '(' token
 #define PORT              ((Port     *)0x41008000UL) /**< \brief (PORT) APB Base Address */
                           ^

......

                              ^
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:449:4: note: in expansion of macro 'DONE'
    DONE;
    ^
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:192:10: error: expected ';' before ']' token
     [PORT] "M" (FastPin<DATA_PIN>::port()-0x20),  \
          ^
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:278:36: note: in expansion of macro 'ASM_VARS'
 #define DONE asm __volatile__("2:" ASM_VARS );
[... 11 lines cut ...]
                 from /var/folders/ns/936wknzn38x4vjh0l3z0sj740000gn/T/arduino_modified_sketch_375437/Blink.ino:1:
/Users/bracero/Documents/Arduino/libraries/FastLED/fastpin.h: In instantiation of 'class FastPin<3u>':
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:96:49:   required from 'class ClocklessController<3u, 30, 75, 45, (EOrder)66u, 0, false, 10>'
/Users/bracero/Documents/Arduino/libraries/FastLED/chipsets.h:468:7:   required from 'class WS2812Controller800Khz<3u, (EOrder)66u>'
/Users/bracero/Documents/Arduino/libraries/FastLED/FastLED.h:94:34:   required from 'class NEOPIXEL<3u>'
/Users/bracero/Documents/Arduino/libraries/FastLED/FastLED.h:310:28:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = NEOPIXEL; unsigned char DATA_PIN = 3u]'
/var/folders/ns/936wknzn38x4vjh0l3z0sj740000gn/T/arduino_modified_sketch_375437/Blink.ino:9:56:   required from here
/Users/bracero/Documents/Arduino/libraries/FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^
In file included from /Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/fastled_avr.h:6:0,
                 from /Users/bracero/Documents/Arduino/libraries/FastLED/platforms.h:32,
                 from /Users/bracero/Documents/Arduino/libraries/FastLED/FastLED.h:55,
                 from /var/folders/ns/936wknzn38x4vjh0l3z0sj740000gn/T/arduino_modified_sketch_375437/Blink.ino:1:
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h: In instantiation of 'void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showPixels(PixelController<RGB_ORDER>&) [with unsigned char DATA_PIN = 3u; int T1 = 30; int T2 = 75; int T3 = 45; EOrder RGB_ORDER = (EOrder)66u; int XTRA0 = 0; bool FLIP = false; int WAIT_TIME = 10]':
/var/folders/ns/936wknzn38x4vjh0l3z0sj740000gn/T/arduino_modified_sketch_375437/Blink.ino:21:1:   required from here
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:112:7: error: 'cli' was not declared in this scope
   cli();
       ^
/Users/bracero/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:165:7: error: 'sei' was not declared in this scope
   sei();
       ^
exit status 1
Error compiling for board Adafruit ItsyBitsy M4 (SAMD51).
```

focalintent, 2018-08-18:
If the SAMD51 is effectively the same as the SAMD21 then there just needs to be a platform/pin definitions for the ItsyBitsy M4 added to https://github.com/FastLED/FastLED/blob/master/platforms.h and https://github.com/FastLED/FastLED/blob/master/led_sysdefs.h and https://github.com/FastLED/FastLED/blob/master/platforms/arm/sam/fastpin_arm_sam.h -- if the SAMD51 is significantly different in any way, then the library will be a bit more porting work (as loosely described in https://github.com/FastLED/FastLED/blob/master/PORTING.md but may be able to use the samd21 code as a reference).

vccabral, 2018-08-20:
@focalintent, I am reading the guide and attempting to work on a port. I want to test out the existing SAMD21 code base.

When the porting.md guide instructs the developer to "conditionally include platform sysdefs header file". 

1. Are these the locations I need to conditionally include platforms?
https://github.com/FastLED/FastLED/blob/master/led_sysdefs.h#L24-L25
https://github.com/FastLED/FastLED/blob/master/platforms.h#L24-L25

2. What are those variables (shown below) and how can I find the one I should include on my itsy bitsy 

```#elif defined(__SAMD21G18A__) || defined(__SAMD21J18A__) || defined(__SAMD21E17A__) || defined(__SAMD21E18A__)
```

Thanks for the help.

focalintent, 2018-08-20:
I’ve found finding them to be a bit of a black art and mostly do it by trial and error and header file digging.

vccabral, 2018-10-03:
@oscarmiike no luck so far. I put FastLED and used circuit python. I would love to use FastLED though. I will recompile my old efforts to see that final state if interested.

closed 2018-11-21 by focalintent