FastLED/FastLED #641: Brightness floor not functional with color correction.

7lbmonkey, 2018-08-17:
Using color correction at low light levels results in some led's turning off entirely, rather then staying floored. A white light dimmed down would become yellow, then red.
I'm unfamiliar with the library, but I'm looking for where color correction is actually applied?

focalintent, 2018-08-18:
The color correction is basically just an extra level of scaling applied to the output values - as the color correction itself is expressed as a scaling operation - and right now the scaling floors to 0 not 1 (in part to avoid having the overhead of checking for a non-zero start, e.g. for the asm optimized code where I'm out of cycles on some platforms - but also because we made a conscious choice to have the brightness floor to black, not white - for example if you have a gradient across a strip from 0 to 255, we didn't want lowering the brightness to bring the strip down to a single level of color until flipping to black at the last step from 1 to 0) - this is an issue even even without color correction (as the scaling can bring some components of a color down to 0 faster than others when using colors other than white).

There really isn't a single place to easily fix this - some controllers will scale using the loadAndScale methods in https://github.com/FastLED/FastLED/blob/master/controller.h -- in there you can probably replace the scale8 calls with scale8_video calls (scale8_video floors to 1, not 0) -- the ones implemented in asm, basically have it all unrolled into the asm code, interleaved with other work, and would be more difficult to switch to scale8_video (especially the AVR asm code, where there isn't really the cycle overhead available to add the extra conditional checks)

Either way - whether scaling floors at 0 or 1, though, for very very low values, you're going to see your color temperature shift as the brightness values get lower and lower - simply because the "range of possible values" narrows to the point where you can't get the rgb ratios that you want and it will either flatten to a temperature of {255,255,255} (if you change scaling to floor to 1 instead of 0) or individual channels zero out (the current behavior).

In an ideal world, we would've all moved to leds that use 16-bit brightness values instead of 8-bit - but I have no say over the hardware that's out there : )

7lbmonkey, 2018-08-18:
Thanks for the explaination, I appreciate it. I added my own color correction on the front end, and realized you were right to make it floor to 0. It looked bad, but not nearly as distracting as flooring to 1. I got results I was happy with by just clipping off every color below the ratio I wanted and filling with black.
And thanks again for such a fast and helpful response!