FastLED/FastLED #646: Fastled 3.2.0 seg fault during linking

nellocriscuolo, 2018-08-29:
Hi All,
I'm compiling a simple Blink example with more then 6 row like this:
`  FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);`
During linking I got an error in ShowPixels function and the compiling ended with segmentation fault.
The same example compile with success with FastLed 3.1.6.

What wrong in 3.2.0?

Best Regards
N

focalintent, 2018-08-29:
I need a little bit more information - what version of Arduino are you building with, and what hardware are you building for? 

Also sharing both the exact code you’re compiling and the error you are getting would be helpful.  

Don’t make me have to guess all of this stuff. I

NelloCris, 2018-08-30:
Arduino version: 1.8.6
Hardware: Arduino Pro or Pro Mini with Atmega328p

These are the data pins:

```
  #define DATA_PIN1 6
  #define DATA_PIN2 7
  #define DATA_PIN3 8
  #define DATA_PIN4 9
  #define DATA_PIN5 12
  #define DATA_PIN6 13
  #define DATA_PIN7 A4
  #define DATA_PIN8 A5
```

In setup() I have 
```
       FastLED.addLeds<WS2811, DATA_PIN1, RGB>(leds, NUM_LEDS);
       FastLED.addLeds<WS2811, DATA_PIN2, RGB>(leds2, NUM_LEDS);
       FastLED.addLeds<WS2811, DATA_PIN3, RGB>(leds, NUM_LEDS);
       FastLED.addLeds<WS2811, DATA_PIN4, RGB>(leds2, NUM_LEDS);
       FastLED.addLeds<WS2811, DATA_PIN5, RGB>(leds, NUM_LEDS);
       FastLED.addLeds<WS2811, DATA_PIN6, RGB>(leds2, NUM_LEDS);
       FastLED.addLeds<WS2811, DATA_PIN7, RGB>(leds, NUM_LEDS);
       FastLED.addLeds<WS2811, DATA_PIN8, RGB>(leds2, NUM_LEDS);
```

 The error I got:

```
C:\Users\nello\Documents\Arduino\libraries\FastLED/platforms/avr/clockless_trinket.h: In member function 'showPixels':

C:\Users\nello\Documents\Arduino\libraries\FastLED/platforms/avr/clockless_trinket.h:167:2: internal compiler error: Segmentation fault

  }

  ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: E:\arduino-1.8.6\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

e:/arduino-1.8.6/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

Uso la libreria FastLED alla versione 3.2.0 nella cartella: C:\Users\nello\Documents\Arduino\libraries\FastLED 
exit status 1
Errore durante la compilazione per la scheda Arduino Pro or Pro Mini.
```

If I comment the last two rows of addLeds the program link successfully:
```

       FastLED.addLeds<WS2811, DATA_PIN1, RGB>(leds, NUM_LEDS);
       FastLED.addLeds<WS2811, DATA_PIN2, RGB>(leds2, NUM_LEDS); //FA LO STESSO DEL PIN 1
       FastLED.addLeds<WS2811, DATA_PIN3, RGB>(leds, NUM_LEDS);
       FastLED.addLeds<WS2811, DATA_PIN4, RGB>(leds2, NUM_LEDS);
       FastLED.addLeds<WS2811, DATA_PIN5, RGB>(leds, NUM_LEDS);
       FastLED.addLeds<WS2811, DATA_PIN6, RGB>(leds2, NUM_LEDS);
//       FastLED.addLeds<WS2811, DATA_PIN7, RGB>(leds, NUM_LEDS);
//       FastLED.addLeds<WS2811, DATA_PIN8, RGB>(leds2, NUM_LEDS);
```

NelloCris, 2018-08-30:
Just copied the same code in other file and it works. I think the problem is only mine. Thanks.

yesyesuk, 2018-08-30:
A4 and A5 are the I2C pins. Are you using I2C / Wire somewhere else in your sketch?

NelloCris, 2018-08-30:
No. Sometimes it compiles and links well and sometimes not...

Adrianotiger, 2018-10-01:
I had it on different sketches using Arduino 1.8.6. Now, with Arduino 1.8.7 it seems to be solved. But has nothing to do with this library (because I had it on many different sketches using different libraries).

rosenbjerg, 2018-11-17:
I still experience the linking problem on v. 1.8.7 (Windows Store 1.8.15.0)

focalintent, 2018-11-18:
Ultimately, this is a bug in the compiler, it sounds like.

maxnz, 2018-11-29:
In my testing, I found that 1.8.5 seems to work just fine but 1.8.6, 1.8.7 and Microsoft Store 1.8.7 throw seg faults. Exactly the same error as @NelloCris in all three as far as I can tell. My friend also ran into this same problem on MS 1.8.7 too.

focalintent, 2019-01-20:
Going to call this an Arduino bug then, for now and close this off here.

closed 2019-01-20 by focalintent