FastLED/FastLED #649: Brightness control CHSV vs CRGB

MudiStar, 2018-09-02:
Hi all,

I recognized a deviation when I want to increase the brightness of the red LED from a WS2812B, COLOR_ORDER GRB.

The program increase the brightness from 0 to e.g. 30 from two LEDs next to each other:
leds[0]=CHSV(0,255,brightness)
leds[1]=CRGB(brightness,0,0)

When I do this, I found that LED "0" stays the same brightness until brightness=16.
Then the brightness makes a "jump", but even then, the LED "0" is still darker then LED "1".
So the input for both is 16 but the brightness is different.

I didn't specify a color temperature.

The goal is to design a sunrise from red to yellow and at the same time from dark to bright according to the logarithm of the human eye. I want to have the sunrise as smooth as possible. This is why I found this issue.
16-Bit PWM would be perfect for the WS2812B (I made good experience with some other hardware), so I know that there are physical limits with the 8-Bit.

Thanks for your help

focalintent, 2018-09-04:
HSV conversion in FastLED does its own gamma adjustment, which is likely what you are seeing.  Mark can speak more to the specifics of what’s happening here...

MudiStar, 2018-09-07:
Ok thanks for your answer.
Is there a way to avoid this?

focalintent, 2018-09-09:
My recommendation to folks is to generally work entirely in either CHSV or CRGB so that things stay consistent.

kriegsman, 2018-09-09:
You are also welcome to clone the hsv2rgb function... but leave out the brightness correction at the end, which might give you a curve like the one to think you might want.

MudiStar, 2018-09-11:
@focalintent 
I was working only with the CHSV until I recognized the problem

Ok thanks for your support 
I will try to work with the CRGB function

closed 2018-09-11 by MudiStar