FastLED/FastLED #651: ESP32 and WS2812b are not working

jorgecis, 2018-09-09:
I have a 1m WS2812B strip, runnig the following code in my ESP32 dev, only the first turn on Green all the time, so it never go off, sometimes the brightness is a little different but never go off.

```
#include "FastLED.h"
#define NUM_LEDS 30
CRGB leds[NUM_LEDS];
void setup() { FastLED.addLeds<NEOPIXEL, 12>(leds, NUM_LEDS); }
void loop() {
	leds[0] = CRGB::White; FastLED.show(); delay(30);
	leds[0] = CRGB::Black; FastLED.show(); delay(30);
}
```

   I tried all the other examples, most of the time only the first led turn on/flick in ramdom colors. I tried with the pins 12,13,14 all give me same results.

   Using the same wires setup and a ESP8266, all the examples works fine. The only change in the code betweeen boards are the GPIO PIN.

  Any ideas?

focalintent, 2018-09-09:
What version of the library do you have I staled?  Also, do you ha e the most recent esp32 platform installed?  Finally, do you have a level shifter between the esp32 and the leds?

jorgecis, 2018-09-09:
Hi, Thanks for taking the time to check this issue.

   I was using the last version of everything (ESP32 core, libs, ..), but I found the problem and is not in your code, in many forums and website like this one https://learn.adafruit.com/adafruit-neopixel-uberguide/powering-neopixels it said that we need to install a resistor in the data pin.  

> Adding a 300 to 500 Ohm resistor between your microcontroller's data pin and the data input on the first NeoPixel can help prevent voltage spikes that might otherwise damage your first pixel. Please add one between your micro and NeoPixels!

  So I did that, I added a 470 Ohm resistor and that was the cause of my issues, without the resistor and using the pin 25 all the examples are working fine, the animations are smooth. 

   So the ESP8266 works fine with and without the resistor but the ESP32 only works without the resistor. 

   I'm not sure if the problem is only with my led strip ( I have only one).

marcmerlin, 2018-12-30:
@jorgecis  note that @focalintent  said level shifter, to move the voltage up from 3.3V to 5V. Adding a resistor actually lowers the voltage further.
Adafruit does recommend them, but I've never used them and I've had strips where adding them made the strip not work.
So yeah, don't use resistors.
Do you mind resolving/closing the issue?

jorgecis, 2018-12-30:
Is not an issue with the library so I closed this issue

closed 2018-12-30 by jorgecis