FastLED/FastLED #656: APA102/SK9822 global brightness sometimes incorrect

cmsavage, 2018-09-26:
I am very happy that FastLED is now making use of the global brightness aspect of the APA102/SK9822 LEDs.  Even with the brightness turned down very low (and using much less power), these LEDs are capable of providing a wide dynamic range of colors with this feature.

However, while trying out the global brightness setting for an SK9822 LED strip, I noticed some odd brightness behavior that I was able to track down the cause of.  Specifically, in the following _[chipsets.h](https://github.com/FastLED/FastLED/blob/38105c61de097066946f2df4d2bb2c36ddc78d7c/chipsets.h#L193-L201)_ code:
```
#if FASTLED_USE_GLOBAL_BRIGHTNESS == 1
                const uint16_t maxBrightness = 0x1F;
                uint16_t brightness = (max(max(s0, s1), s2) * maxBrightness >> 8) + 1;
                s0 = ((uint16_t)s0 + 1) * maxBrightness / brightness - 1;
                s1 = ((uint16_t)s1 + 1) * maxBrightness / brightness - 1;
                s2 = ((uint16_t)s2 + 1) * maxBrightness / brightness - 1;
#else
                const uint8_t brightness = 0x1F;
#endif
```
the right side of the `s0` calculation is occasionally greater than 255, in which case `uint8_t s0` wraps around (262 → 6), resulting in a grossly incorrect brightness.  This happens for several values of `s`.

Even if the overflow is caught and `s` is set to 255, the algorithm does not preserve scale\*brightness, leading to significant deviations in the resulting LED brightness at lower scale factors.  I suggest using the following, in which scale\*brightness deviates by at most 0.4% (and typically much less):
```
#if FASTLED_USE_GLOBAL_BRIGHTNESS == 1
                const uint16_t maxBrightness = 0x1F;
                uint16_t brightness = ((((uint16_t)max(max(s0, s1), s2) + 1) * maxBrightness - 1) >> 8) + 1;
                s0 = (maxBrightness * s0 + (brightness >> 1)) / brightness;
                s1 = (maxBrightness * s1 + (brightness >> 1)) / brightness;
                s2 = (maxBrightness * s2 + (brightness >> 1)) / brightness;
#else
                const uint8_t brightness = 0x1F;
#endif
```
The above algorithm does not experience the `uint8_t` overflow.

cmsavage, 2018-09-26:
Attaching spreadsheet for looking at brightness and scale calculations.

[GlobalBrightness_rollover.xlsx](https://github.com/FastLED/FastLED/files/2418737/GlobalBrightness_rollover.xlsx)

extesy, 2018-09-26:
I didn't have SK9822 to test this on but I do have it now and will check the math soon.

extesy, 2018-09-27:
One thing I've noticed during the testing is that when `setCorrection(TypicalLEDStrip)` is used (like in provided FastLED examples) the hue shift is very apparent on low brightness settings. White color becomes some weird purple-ish tint.

christophepersoz, 2018-09-27:
Indeed I also noticed that those days as I switch my led strip from WS2813 to APA102C but I haven’t had the time to find the source of the problem.

cmsavage, 2018-09-27:
I have been using `setCorrection(UncorrectedColor)` and did not notice a hue change through even the lowest brightness levels.  Now that I try `setCorrection(TypicalLEDStrip)`, I can see the hue changes at very low brightness changes.  I can see one issue that causes hue changes: the 8-bit storage of the RGB scale factors in the `controller.h` [computeAdjustment()](https://github.com/FastLED/FastLED/blob/38105c61de097066946f2df4d2bb2c36ddc78d7c/controller.h#L150-L170) routine.  The scale factors `s0, s1, s2` that are used in the APA102/SK9822 global illumination calculation in the parent post above are given by `scale*(Correction[k]+1)*(Temp[k]+1)/65536` where `scale` is the overall scale passed into `show()`, and `Correction` and `Temp` are the RGB color correction and color temperature factors.  Here are some results using the `TypicalLEDStrip` correction:

Scale | Correction | Adjusted `(s0,s1,s2)`
--------|--------|--------
255 | (255,176,240) | (255,176,240)
128 | (255,176,240) | (128,88,120)
64 | (255,176,240) | (64,44,60)
32 | (255,176,240) | (32,22,30)
16 | (255,176,240) | (16,11,15)
8 | (255,176,240) | (8,5,7)
4 | (255,176,240) | (4,2,3)
3 | (255,176,240) | (3,2,2)
2 | (255,176,240) | (2,1,1)
1 | (255,176,240) | (1,0,0)

Because the final scale factors are restricted to integer values, they cannot be kept in the same ratios as the numbers get small.  Calling `show(4)` results in a more purple hue [→(4,2,3)] than `show(255)` [→(255,176,240)], while `show(3)` is reddish [→(3,2,2)], `show(2)` is even more red [→(2,1,1)], and `show(1)` is just red [→(1,0,0)].

This scaling procedure works fine for 8-bit RGB outputs, but APA102/SK9822's essentially have a dynamic range of 31*256 (~13 bits) that this procedure was not meant to handle (in its current form).

@extesy, @christophepersoz: When you say low brightness levels, are you talking about scales below around 8, where the finite resolution effects of the adjusted scales starts to be significant?

christophepersoz, 2018-09-27:
Thanks for all this explanations. 
What I did notice apply to all the brightness scale value corrected. The APA102C tends to be more reddish when setCorrection is applied. 
I haven’t had the time to investigate more about this as I’m currently abroad for work. 
Keeping you in touch when back home.

PR #657 (2019-01-20): Update global brightness math to fix overflows
Thanks to @cmsavage for a heads up. I was able to reproduce the issue and applied the suggested fix. With the fix I don't see overflows happening anymore. Fixes #656.

/cc @focalintent
closed 2019-01-20 by focalintent