FastLED/FastLED #659: led flickering (not ESP!)

Adrianotiger, 2018-10-01:
Hi,
I had a neopixel project and rewrote it for FastLED. With this library the leds are flickering.
I am using a simple Arduino mini (16MHz/5V) without any interrupts and ground is connected correctly. As said, it works perfectly with Neopixel library. It happens when I have multiple strips and the LED colors are on the minimum. 
Here the code:

```
#include <FastLED.h>

CRGB ledsStrip[14];
CRGB ledsCircle[24];
CRGB ledsWheel[19];

void setup() {
  FastLED.addLeds<NEOPIXEL, 7>(ledsStrip, 14);
  FastLED.addLeds<NEOPIXEL, 8>(ledsCircle, 24);
  FastLED.addLeds<NEOPIXEL, 9>(ledsWheel, 19);
}

void loop() {
  for(int i=0;i<14;i++) ledsStrip[i] = CRGB(2, 0, 6);
  for(int i=0;i<24;i++) ledsCircle[i] = CRGB(0, 1, 1);
  for(int i=0;i<19;i++) ledsWheel[i] = CRGB(1, 1, 0);
  FastLED.show();
  delay(1);
}
```
Maybe the timings is not like expected or the ws2812b leds are not originals. But it is strange that it works with Neopixels and I see the flickering with the FastLED library.

focalintent, 2018-10-01:
Try turning off dithering - it’s possible that is what you are seeing at such low brightness levels.

Adrianotiger, 2018-10-01:
Thank you. 
With `FastLED.setDither(0)` it works.

I was reading this:
https://github.com/FastLED/FastLED/wiki/FastLED-Temporal-Dithering
And if I understand it correctly, it should be activated only if brightness is not 255.

closed 2019-01-20 by focalintent