FastLED/FastLED #663: Curious behavior with FastLED.show()

moderndevice, 2018-10-07:
In the code below, if I move FastLED.show() between functions, the functionality breaks.
I've spent a long time debugging but ultimately have no idea where to look next. Any ideas would be welcome as to where the issue could be.

Details - using a Atmega368 (Arduino Uno) type board with some Neopixel type LED strips.

#include "FastLED.h"

// How many leds in your strip?
#define NUM_LEDS 77
#define FirstLEDsStart 0
#define FirstLEDsEnd 36 // small gap before next pattern begins

#define SecondLEDsStart 38
#define SecondLEDsLength 36
#define stripMSdelay 40

void fillStripesFromPalette(int blockSize);

unsigned long lastStripeTime;

// What pin are you using for data (in series with 470 ohm resistor)?
#define DATA_PIN 7
// #define CLOCK_PIN 13

// Define the array of leds
static CRGB leds[NUM_LEDS];

CRGB colorP[] =
{ CHSV(0, 255, 255), CHSV(220, 250, 150), CHSV(50, 250, 150),
  CHSV(120, 250, 120), CHSV(40, 200, 50), CHSV(200, 150, 90),
  CHSV(55, 250, 75), CHSV(70, 200, 150), CHSV(240, 150, 150),
  CHSV(150, 250, 200)
}; // add as many as you like - nothing should break!
// color palette not used in this demo!iu9

int maxColorIndex = sizeof(colorP) / sizeof(colorP[0]);

void setup() {
  Serial.begin(9600);

  // FastLED.addLeds<WS2811, DATA_PIN, RGB>(leds, NUM_LEDS);
  // FastLED.addLeds<WS2812, DATA_PIN, RGB>(leds, NUM_LEDS);
  // FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, NUM_LEDS);
  FastLED.addLeds<NEOPIXEL, DATA_PIN> (leds, NUM_LEDS);

  randomSeed(analogRead(A6));
}

void loop() {
  randPixels(FirstLEDsStart, FirstLEDsEnd);              // keep both functions running

  if (millis() - lastStripeTime > stripMSdelay) {  // timing for fillStripesFromPalette function
    fillStripesFromPalette(SecondLEDsStart, SecondLEDsLength, 15) ; // function  running on a separate array
    lastStripeTime = millis();

  }

  FastLED.show(); // is in the fillStripesFromPalette function  - there is a bug I couldn't find

}

void randPixels(int startPix, int endPix) {    // fills strip with random pixels
  // keeps track of pixels chosen and chooses all pixels
  // before repeating again - maybe not worth the hassle?
  // Uses random walk for colors.

  static CHSV color = CHSV(0, 200, 150);  // assign starting color

  color.hue = color.hue + (random(17) - 8); // hue random walk
  color.sat = color.sat + (random(7) - 3); // saturation random walk
  color.value = color.value + (random(17) - 8); // value random walk

  // how to contstrain values to a reduced range
  color.sat = constrain(color.sat, 80, 255);   // uncomment to view
  // color.value = constrain(color.sat, 80, 255);   // uncomment to view

  int randPix = random(endPix - startPix) + startPix; // find a random pixel in our chosen range

  // compare the random walk above with
  // the completely random pixel line below. Uncomment to view.
  // color =   CHSV(random(255), random(255), random(255)); // totally random color
  leds[randPix] =  color;  // random walk - write the pixel
}

void fillStripesFromPalette(byte startPixel, byte LengthPixels, int blockSize) {
  // fills the area with travelling random sized strips
  // colors of strips are derived from the palette above
  // sizes of strips determined with blockSize

  //blocksize determines number of max pixel "strip width"

  byte stripeWidths[maxColorIndex]; // using palette array for number
  // of widths so the arrays correspond one-to-one.
  static byte initialized = 0;
  static byte offset;
  static CRGB pixColor;
  static byte colorIndex;           // counter to keep track of stripe widths
  //                                // and colors

  if (!initialized) { // fill up stripe width array
    // but only do it the first time function is called
    for (int i = 0; i < maxColorIndex; i++) {
      stripeWidths[i] = random(1, blockSize); // random stripWidths
    }
    initialized = 1;    // set flag so this only happens once
  }

  offset++;
  colorIndex = 0;

  for (byte i = startPixel; i < (startPixel + LengthPixels); ) { // note - no increment

    // chose stripe width and new color
    colorIndex = (colorIndex + 1) % maxColorIndex; // controls
    // both palette array and stripeWidth array
    byte pixelsInStripe = stripeWidths[colorIndex] ;
    pixColor = colorP[colorIndex];

    for (int j = 0; j < pixelsInStripe; j++) {  // fill one stripe

      leds[((i + offset) % LengthPixels)  + startPixel] = pixColor;  // write the pixel
      i++;  // note that loop counter is from outside loop
      if ( (byte)i >= ( (byte)startPixel +  (byte)LengthPixels)) {
        break;
      }
    }

    FastLED.show();   // this should really be in the loop, there is some kind of
    //                 // bug with FastLED.show() that I haven't found - so it works correctly here
  }
}

moderndevice, 2018-10-07:
> It’s possible that when you have FastLED.show() in loop it is running so frequently that the timer is never getting a chance to increment. You don’t really describe how things aren’t working.
>
> Try putting a delay(10); after the call to FastLED.show() in the loop as an experiment.

Thanks for that.

What happens is that the strip that normally does a series of moving stripes, displays a single color.
I haven’t really looked to see what color it might be. (colors come from a CRGB array)

Thanks much for the feedback

moderndevice, 2018-10-07:
I tried it but no joy.
Kind of mysterious eh?

Paul Badger

marmilicious, 2018-10-08:
@moderndevice  My suspicion is you're trying to write pixel data to a pixel that doesn't exist (ie outside of the range of NUM_LEDS) in the fillStripesFromPalette function.

closed 2019-08-12 by focalintent