FastLED/FastLED #664: AnalogOutput brightness is 8bit, but seems to need 10bit

demlak, 2018-10-08:
Hi,
when i play with this example: https://github.com/FastLED/FastLED/blob/master/examples/AnalogOutput/AnalogOutput.ino
On my Wemos D1 Board, the brightness is not at the maximum.

I mesured the voltage o the output. it is at ~0,84V instead of 3,3v that it should have.

well.. 0,84v is 1/4 of 3,3v
and 256 is 1/4 of 1024

so it seems to me, that the wemos d1 has a 10bit output, but fastled just puts 8bit to the analog pins. 

any idea on how to fix this?

focalintent, 2019-01-03:
FastLED is currently 8 bit internally - you could, for now, multiply the values by 4 before writing them out in the analog outout example (just make sure you put the values in an int so that it doesn’t get cut off at 8 bits).  (In the future FastLED will be either 8 or 16 bit internally, which will make it more trivial to support 10/12/16 bit outputs directly).

CyanoFresh, 2019-01-03:
yep, but multiplying is not smooth for rainbow animation

demlak, 2019-01-03:
8bit is not smooth for anything.

btw. i canceled the use of FastLED for my Project. And i bought a PCA9685 to get smooth 12bit output for my analog LED Stripes. I will see, what software i will use to drive this.

chrisparton1991, 2019-01-03:
@demlak I've found 8 bits to be smooth enough for my use cases.

Also, please keep in mind that @focalintent has given us countless hours of his time and effort for free, so politeness is the least we can offer in return :)

demlak, 2019-01-03:
I did not and will not discredit any work of any fastled member

focalintent, 2019-01-04:
He’s right - 8bit is pretty poor - for my own work I do all my higher level color work in 16bit and then convert down to 8-bit hsv/rgb at the last moment.  This is why I’m adding support for 16-bit rgb and hsv pixels in addition to the current 8-bit CRGB/CHSV as part of the RGBW rewrite - they will convert down to 8/10/12 bit as necessary depending on your output device.

marmilicious, 2019-01-06:
Oooh, that sounds excellent @focalintent

focalintent, 2019-01-20:
This'll happen by default once FastlED supports > 8 bit color/output work.

closed 2019-01-20 by focalintent