FastLED/FastLED #669: single WS2812 need to skip indexes to get address right

sylvix, 2018-10-18:
Hi! 

RGB modules are:
3 x FC-100 WS2812 (presumably) single board LEDs. They look like this: 
[image]

Also, it's searchable by "FC-100 RGB" on Google. In one place there's info that it's WS2811, other place it's WS2812.

Main setup:
Arduino Uno
Power comes from Arduino 5V and GND pins
Data wire is connected to pin 3
There is a 470 uF capacitor between 5V and GND (the only one I had), data line has 470 ohm resistor.
Everything is connected with wires on a breadboard chain-like, looking like that:

[image]

Only my wires are longer, using prebuilt wires for arduino and breadboard (male-male).

Using FastLED 3.2.1, Arduino IDE 1.8.7 on Win10, nothing else.

When I use 1 LED, everything works perfectly, i.e. blink:

```
#include <FastLED.h>
#define NUM_LEDS 1
#define DATA_PIN 3
CRGB leds[NUM_LEDS];

void setup() {
    delay(2000);
    FastLED.addLeds<WS2812, DATA_PIN, RGB>(leds, NUM_LEDS);
}

void loop() {
  leds[0] = CRGB(255, 0, 0);
  FastLED.show();
  delay(500);
  leds[0] = CRGB(0, 0, 0);
  FastLED.show();
  delay(500);
}
```

Next, I'm adding one module. Trying to blink with two LEDS:

Changing
```#define NUM_LEDS 2```

Adding coloring to loop:
```
void loop() {
  leds[0] = CRGB(255, 0, 0);
  leds[1] = CRGB(0, 255, 0);
  FastLED.show();
  delay(500);
  leds[0] = CRGB(0, 0, 0);
  leds[1] = CRGB(0, 0, 0);
  FastLED.show();
  delay(500);
}
```
Now only the first one blinks, the second one is blue (default color when I power it on even without data line).

Ok, now the strange thing:

Increased NUM_LEDS to 3:
```#define NUM_LEDS 3

void loop() {
  leds[0] = CRGB(255, 0, 0);
  leds[2] = CRGB(0, 255, 0);
  FastLED.show();
  delay(500);
  leds[0] = CRGB(0, 0, 0);
  leds[2] = CRGB(0, 0, 0);
  FastLED.show();
  delay(500);
}
```
NOW it somehow works, but sometimes it goes off at random (blue, purple colors instead of turning off). So the first LED works correctly as ```leds[0]```, second LED works as ```leds[2]```, and ```leds[1]``` does nothing. If I connect third LED to the chain, set ```NUM_LEDS``` to 5 and use it as ```leds[4]```, it also kinda works.

Can anyone tell me what's the problem? Are my LEDs faulty? Or is timing wrong for those specific LEDs? I'm sure it's not normal to double array size and skip indexes by one to get the address right.

lanternfish, 2018-11-05:
My understanding of these modules are the packets are sequential i.e. packet 1, packet 2 ... so the chips receive the data, check for end of data and if not end of data then pass on packet and receive next packet. Thus with three modules they will always receive the last n packets. If you send 5 packets but have only three modules then they will use packets 3,4,5. Correct?
A thought: breadboards (especially cheaper ones) can be problematic with making a good connection between pin and contact (oxidisation, grime ...). Try reinserting modules a number of times.
A question: Have you swapped the modules around to see if one is faulty?

sylvix, 2018-11-06:
Nope, this is not what happens. It looks like one module "eats" two packets at once. My code demonstrates that - first module lights up on modifying 0-th index of pixel array, second one lights up on modifying 2-nd index, third one - 4-th index, fourth - 6-th index, etc.
They are connected very tightly, one after another, I tried resoldering, another wires, so I think it's not a problem with breadboard.

lanternfish, 2018-11-06:
Timing issue? WS2811 and WS2812 have slightly different timing requirements and the WS2811 has a high speed mode which halves high and low timings (but not the reset).
Have y0u tried setting FastLED to WS2811?
`FastLED.addLeds<WS2811, DATA_PIN, RGB>(leds, NUM_LEDS);`

focalintent, 2018-11-07:
I noticed that the leds are 6-pin, WS2812's are only 4 pin - I wonder if what happened is that they put WS2813's on there, but wired up DIN to the backup input pin (the way the WS2813 works is if it gets data on its main DIN pin, it displays the first 24-bits of data that it gets then passes the rest along, if it doesn't get data on that pin, but it does get data on the backup pin, it will skip 24 bits of data, then display the next 24 bits of data that it gets, and then pass the rest along.  This is how the WS2813 accounts for individual leds dying).

closed 2019-01-20 by focalintent