FastLED/FastLED #670: Serial.println() kills Palette Knife on ESP8266

janfiess, 2018-10-19:
The palette knife for FastLED actually works as expected with a WS2812B strip, but as soon as I add a Serial.println() to the code, the strip won´t change the color anymore. It just keeps the previous color.
Is there a workaround?

`
#define FASTLED_ALLOW_INTERRUPTS 0
#include "FastLED.h"

DEFINE_GRADIENT_PALETTE( ib_jul01_gp ) {
    0,   0,  0,  0,
   64,   111,111,  0,
  128,   0,  0,  0,
  172,   0,200, 200,
  255,   0,  0,  0
};

#if FASTLED_VERSION < 3001000
#error "Requires FastLED 3.1 or later; check github for latest code."
#endif

#define DATA_PIN    2
#define LED_TYPE    WS2812B
#define COLOR_ORDER GRB
#define NUM_LEDS    50
#define BRIGHTNESS  100
#define DELAY       25

CRGB leds[NUM_LEDS];
static const uint8_t interval = 0;

void setup() {
  delay(3000); // 3 second delay for recovery
  Serial.begin(9600);
  
  // tell FastLED about the LED strip configuration
  FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS)
    //.setCorrection(TypicalLEDStrip) // cpt-city palettes have different color balance
    .setDither(BRIGHTNESS < 255);

  // set master brightness control
  FastLED.setBrightness(BRIGHTNESS);
}

CRGBPalette16 gTargetPalette( ib_jul01_gp );

void loop()
{
  sunrise(); 
  FastLED.show();
  FastLED.delay(DELAY);
}

void sunrise() {
  // current gradient palette color index
  static uint8_t heatIndex = 0; // start out at 0

  CRGB color = ColorFromPalette(gTargetPalette, heatIndex);

  // fill the entire strip with the current color
  fill_solid(leds, NUM_LEDS, color);

  EVERY_N_SECONDS(interval) {
    // stop incrementing at 255, we don't want to overflow back to 0
    if(heatIndex < 255) {
      heatIndex++;
    }
    if(heatIndex >= 255) {
      heatIndex = 0;
    }
  }
}
`

focalintent, 2018-10-25:
try putting the println in an EVERY_N_MILLISECONDS -- e.g. 

```
EVERY_N_MILLISECONDS(250) { 
  Serial.println("Hello");
}
```

It's also possible that the combination of calling FastLED.show(), FastLED.delay(25) and outputting that bit of serial is tripping things up with the watchdog, but without seeing the output of your crash/serial console it's hard to tell for sure.

(Also - generally speaking troubleshooting questions like this get answered more quickly in the g+ group at http://fastled.io/+ -- more people watching there than here, which is mostly used for tracking library bugs/enhancements)

marcmerlin, 2018-12-30:
@janfiess can you please close/resolve the issue please?

closed 2018-12-31 by janfiess