FastLED/FastLED #672: Fails to compile ATtiny841

YiZhiQi, 2018-10-25:
Trying to compile Blink sketch from FastLED/Examples. Compiling fails.

Testing conditions:
Arduino 1.8.7 (also tried 1.8.5 on another Windows PC)
ATtinyCore 1.2.0 (841@8Mhz internal BOD and LTO disabled)

The only changes in the sketch:
`#define DATA_PIN 3` -----> `#define DATA_PIN 8`
`#define CLOCK_PIN 13` -----> `#define CLOCK_PIN 9`

`FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);`
substituted with
`FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);`

the main problem are multiple assembler errors like:

> C:\Users\Iggy\AppData\Local\Temp\ccUsOYML.s: Assembler messages:
> C:\Users\Iggy\AppData\Local\Temp\ccUsOYML.s:262: Error: illegal opcode or for mcu avr25

I have checked FastLED library files and found out that LIB8_ATTINY is defined properly during compiling.
This is strange because when LIB8_ATTINY is defined, all (or maybe almost) assembler code is redefined to C math and functions.

**here is the full arduino output during compiling:**
[https://gist.github.com/YiZhiQi/faaf0a180c6a8a7511cb84547f247188](url)

Thanks!

focalintent, 2018-10-25:
There’s a bunch of avr asm code that isn’t gated by that macro - only things that use the mul opcode for the most part.  I have a hard time believing that the attiny841 got rid of the or opcode, so something else may be going on.  I won’t be able to look at this until I’m home from work tonight (don’t keep FastLED stuff on the dayjob computers :)

YiZhiQi, 2018-10-26:
@focalintent Thank you. Very much appreciate any help. 
I'm quite new to coding, so even if I understand what is going on, with arduino limited debugging options I can't find the exact place where these calls occur

focalintent, 2018-10-26:
Ok - found the problem - it turns out that LIB8_ATTINY wasn't getting defined for the ATtiny841/441.  I just checked in a fix for that - you can grab it right out of GitHub for now until I push a new release.

commit a3c70f59 (2018-10-26): Fix #672 - we weren't setting LIB8_ATTINY for the ATtiny841/441 - so asm opcodes that don't exist on the tiny cores were getting used...
closed 2018-10-26 by focalintent