FastLED/FastLED #676: ESP32 IRAM interrupthandler

edjeed, 2018-11-07:
The function:
Interrupthander for the ESP32 (using RMT) is placed in IRAM, but calls functions that are not in IRAM.
The result of this is that sometimes errors occur, and the leds are not being updated anymore.

The documentation states:
```
Interrupt handlers must be placed into IRAM if ESP_INTR_FLAG_IRAM is used when registering the interrupt handler. In this case, ISR may only call functions placed into IRAM or functions present in ROM.
```

kriegsman, 2018-11-07:
Thanks for opening this ticket.  What are your thoughts, @samguyer ?

marcmerlin, 2018-12-30:
@edjeed in the meantime did you try changing the driver to use IRAM and does it work better for you? If so, submit a pull request if Sam is too busy to look at this.

commit af74ae7e (2019-01-20): Make sure that everything called from the interrupt handler for ESP32 is marked with  IRAM_ATTR property - should fix #676
closed 2019-01-20 by focalintent