FastLED/FastLED #680: Any updates on Feather M4 support?

AzureEmpyrean, 2018-11-21:
I am working on a project with the Feather M4 and I am unable to complete the project until FastLED is able to be ran on the Feather M4. I would go with another micro controller but the Feather M4 is my best option for running this project right now.

focalintent, 2018-11-21:
Someone submitted pin/port definitions for the itsy bitsy, but no one has yet for the feather m4.  I have one showing up in the next week or two and can poke at it once I have the hardware.  But otherwise, this is a dup of #639

closed 2018-11-21 by focalintent

AzureEmpyrean, 2018-12-04:
Hey focal

> Someone submitted pin/port definitions for the itsy bitsy, but no one has yet for the feather m4. I have one showing up in the next week or two and can poke at it once I have the hardware. But otherwise, this is a dup of #639

Hey I just realized what you said. I found someone that gave me the pin/port definitions for the feather m4 if you havent gotten your feather m4 yet