FastLED/FastLED #681: TM1904 compability

joaopmrod, 2018-11-21:
How can I use FastLED with TM1904 chips? It's similar to WS2812 but with inverted HIGH and LOW. Here is the datasheet: https://www.dropbox.com/s/vmanx0g9pn4p5bs/TM1914A_V1.1_EN.pdf?dl=0

I hope you can help

focalintent, 2018-11-21:
Right now there isn't a good way to do it.  I experimented a few years ago with making it easier to invert the sense of the output line for the clockless chips, but it didn't work well so I mostly removed it.  Since I'm overhauling/re-writing a lot of this code for the RGBW support, I can see about doing the inversion again, but for the most part, I'm tempted to suggest finding a hardware solution to invert the data pin on the fly for you.

focalintent, 2018-11-21:
You can try changing the code - what you specifically want to change depends on what hardware platform you're using - look under the platforms folder in the FastLED library - the controller in question is clockless controller - I found just switching hi and lo for another inverted chipset I was working on ended up not working out quite so well (and in fact blew out a controller testing it because of power draw when the chipset took everything as a 1 and went all white).   Also be aware that the TM1914 has an extra requirement of 6 bytes of frame control data that needs to be sent out at the beginning of every frame that you'd need to add in/account for.

Tobias3sprong, 2020-05-27:
Did you have any luck getting it to work?