FastLED/FastLED #683: Can Parallel Output code really cause LED's to catch on fire?

LongLiveCHIEF, 2018-11-24:
I feel really dumb asking this, but fortunately I have little shame about such things.

On the [Parallel Output](https://github.com/FastLED/FastLED/wiki/Parallel-Output) Wiki, the top has a note:

> N.B. - this code is currently only available on the FastLED3.1 branch and may set your leds on fire!

I'm assuming this is actually a joke?  

I'm somewhat new to electrical engineering, and sometimes it's hard to distinguish between sarcasm and sincerity when I'm missing an inside joke in this field.  I see that there _are_ power limitation functions in this library, so perhaps it's possible?

If it's _not_ a joke, can that warning be expanded on a bit to give some hints about what to avoid? If it _is_ a joke, can we remove it?  Playing with LED's is where a lot of beginners start in this field, and that could really cause someone to shy away from attempting something that is no more inherently dangerous than a normal controller.

kriegsman, 2018-11-24:
It is, in fact, a joke. 
Sorry to have caused concern, and we can definitely add a “(just kidding— your LEDs are safe!)” message or something similar.

LongLiveCHIEF, 2018-11-25:
lol, it's nothing to be sorry for. I assumed correctly, so I don't feel like an idiot. 

I _did_ spend some time digging through code just to make sure there wasn't some inherent power limitation functions that parallel output capable controllers didn't use for some reason.  I'm new to cpp, so the side effect there is that I'm now far more familiar with how this library can be used, and I had to learn about cpp templates in the process, so it's a net win for me.

closed 2019-01-20 by focalintent