FastLED/FastLED #684: support for GS1903

shanosburgess, 2018-11-25:
Hi I received some strips from China with GS1903 IC would you be able to add support or advise if there is already a compatible driver in fastled?

Thanks Shane

szfeco, 2018-12-17:
any progress? I'm facing the same issue...

shanosburgess, 2018-12-17:
Hi mate,

I took some advise from the guy that makes the adrilight software he suggested selecting ws2812b as the chipset, thankfully this has worked for me and I am now able to use the lights.

Let me know how you get on, if I can be of any help I will do what I can.

Cheers and merry Christmas

Shane

szfeco, 2018-12-20:
confirmed. WS2812B works. I tried before not sure was only wiring problem but now works.
Any experience with ws2815 or gs8208 ? i'm planning to order 12V strips invdividually addressable (have some good 12V 400-1200W server PSU-s). Current one 3 leds per address...

closed 2019-01-20 by focalintent

Pfannex, 2019-02-19:
Hi,

for my drawer lights I want to run my GS1903 led strip with the fastLED library. A ESP8266 is used as controller.
Generally the wiring and the simple sketch runs.
The GS1903 is fired with 12VDC, the data pin is directly connected to the ESP8266 GPIO 5. 

I use the work around with the WS2812B model. 
But I´m not able to controll the GS1903 stable.
Sometimes it works sometimes don´t.

For me it looks like communication problems, may be sometimes some bits are missing / wrong? 
Or timing problems?

Can I run some further tests?

greetings 
Pf@nne

commit d50df711 (2019-01-20): Fix #684 - add an alias for GS1903 to the WS2812B timings