FastLED/FastLED #687: FastLED library's delay function isn't reliable

CyborgShadow, 2018-12-02:
https://github.com/FastLED/FastLED/blob/dcbf39933f51a2a0e4dfa0a2b3af4f50040df5c9/wiring.cpp#L85-L95

Using the FastLED library and various examples on an arduino ESP32 with WS2811s, I found inconsistencies.

Given the below scenario:
```code
#define NUM_LEDS 76
#define BRIGHTNESS 254  //brightness  (max 254) 
#define LED_TYPE WS2811
#define LED_PIN 2
#define COLOR_ORDER GRB

#define BRIGHTNESS  200
#define FRAMES_PER_SECOND 60

bool gReverseDirection = false;

CRGB leds[NUM_LEDS];

#define UPDATES_PER_SECOND 100
```

and 2 50 LED strings plugged in, all lights on the string would receive signals, even though only 76 should.

Digging further into the issue with my oscilloscope, it appeared serial data blocks were overlapping at times, not overlapping at others. The blocks were small enough that with the above variables they should not have overlapped (this should occur at 604 LEDs respectively given this configuration).

Tracked it down and it appears that FastLED.delay was causing the issue. While the chip was able to calculate seconds properly using delay, the FastLED.delay was calculating time incorrectly. 

Once I adjusted the FastLED.delay line to use the native delay instead, everything worked perfectly and I started seeing consistent blocks of data on the oscilloscope with equal gaps between them.

focalintent, 2018-12-02:
The purpose of FastLED.delay is to continue driving dithering while delaying - the fact that you were seeing all 100 leds light up instead of just 76 likely means that @samguyer ‘s rmt implementation may not be enforcing a minimum 50us gap between frames.  Sam?

focalintent, 2018-12-02:
FastLED.delay behaves the same way whether dithering is enabled or not.

I suspect that the gapping that you are seeing has to do with something around the RMT timing - waiting for @samguyer to chime in (since he's the one who did that code) on whether or not he's making sure there's enough of a time gap between two frames written out back to back to allow the WS2813's time to reset (and, for that matter, whether or not the RMT code would allow an attempt to write out two overlapping frames at the same time)

samguyer, 2018-12-03:
Let me take a look. At one point, I did have a timer that throttled calls to show() to prevent signals from being too close together.

davepl, 2019-04-30:
If we are not using dithering at all, it is OK to remove FastLED's busy wait/show version of FastLED so that we can still multitask in a friendly manner?

closed 2019-08-12 by focalintent