FastLED/FastLED #690: cant compile fastled for esp32

plankatron, 2018-12-05:
In file included from /home/fleeky/cthuloid/esp32/32wifi12frakfled/32wifi12frakfled.ino:7:0:
/home/fleeky/Arduino/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.008
 #    pragma message "FastLED version 3.001.008"
                     ^
In file included from /home/fleeky/Arduino/libraries/FastLED/FastLED.h:68:0,
                 from /home/fleeky/cthuloid/esp32/32wifi12frakfled/32wifi12frakfled.ino:7:
/home/fleeky/Arduino/libraries/FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from /home/fleeky/cthuloid/esp32/32wifi12frakfled/32wifi12frakfled.ino:9:0:
/home/fleeky/arduino-1.8.5/libraries/PrintLevel/PrintLevel.h:8:57: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     char *gStrings[] = {"silent", "error", "ack", "info"};
                                                         ^
/home/fleeky/arduino-1.8.5/libraries/PrintLevel/PrintLevel.h:8:57: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/home/fleeky/arduino-1.8.5/libraries/PrintLevel/PrintLevel.h:8:57: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/home/fleeky/arduino-1.8.5/libraries/PrintLevel/PrintLevel.h:8:57: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
In file included from /home/fleeky/Arduino/libraries/FastLED/platforms/esp/32/fastled_esp32.h:4:0,
                 from /home/fleeky/Arduino/libraries/FastLED/platforms.h:29,
                 from /home/fleeky/Arduino/libraries/FastLED/FastLED.h:55,
                 from /home/fleeky/cthuloid/esp32/32wifi12frakfled/32wifi12frakfled.ino:7:
/home/fleeky/Arduino/libraries/FastLED/platforms/esp/32/clockless_esp32.h: In instantiation of 'void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::startOnChannel(int) [with int DATA_PIN = 27; int T1 = 60; int T2 = 150; int T3 = 90; EOrder RGB_ORDER = (EOrder)66u; int XTRA0 = 0; bool FLIP = false; int WAIT_TIME = 5]':
/home/fleeky/Arduino/libraries/FastLED/platforms/esp/32/clockless_esp32.h:410:13:   required from 'static void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::startNext(int) [with int DATA_PIN = 27; int T1 = 60; int T2 = 150; int T3 = 90; EOrder RGB_ORDER = (EOrder)66u; int XTRA0 = 0; bool FLIP = false; int WAIT_TIME = 5]'
/home/fleeky/Arduino/libraries/FastLED/platforms/esp/32/clockless_esp32.h:314:26:   required from 'void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showPixels(PixelController<RGB_ORDER>&) [with int DATA_PIN = 27; int T1 = 60; int T2 = 150; int T3 = 90; EOrder RGB_ORDER = (EOrder)66u; int XTRA0 = 0; bool FLIP = false; int WAIT_TIME = 5]'
/home/fleeky/cthuloid/esp32/32wifi12frakfled/32wifi12frakfled.ino:153:1:   required from here
/home/fleeky/Arduino/libraries/FastLED/platforms/esp/32/clockless_esp32.h:432:41: error: 'rmt_register_tx_end_callback' was not declared in this scope
             rmt_register_tx_end_callback(doneOnChannel, 0);
                                         ^
Multiple libraries were found for "WiFi.h"
 Used: /home/fleeky/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /home/fleeky/arduino-1.8.5/libraries/WiFi
exit status 1
Error compiling for board ESP32 Dev Module.

focalintent, 2018-12-05:
@samguyer - any thoughts on what’s missing here?

samguyer, 2018-12-06:
Honestly, I'm not going to be able to look at this issue right now, but I should have more time in the next week or so.

KPhillipps, 2018-12-18:
Any news on this issue, I'm hitting the same thing on an ESP32 Feather.

KPhillipps, 2018-12-18:
I tried to narrow down the issue.  including the library does not result in the error but calling addLeds does.  here is the full program:

#include <FastLED.h>
CRGB leds[1];
#define DATA_PIN 3
#define NUM_LEDS 1

void setup() {
   FastLED.addLeds<WS2812, DATA_PIN, RGB>(leds, 1);
}

void loop() {
}

here is the full error

[image]

focalintent, 2018-12-18:
What’s weird is I have been building go/using esp32’s all week and haven’t seen this issue.

marcmerlin, 2018-12-30:
It definitely works for me too. Is there a chance it's an issue with your arduino environment? Have you tried getting the latest version?  Seems to me like the g++ is different and complains.
Also in preferences, do you have compiler warnings to all? If so, try setting it back to default.

KPhillipps, 2018-12-30:
I got it to work but didn't do methodical enough testing to properly attribute the problem to a specific issue.   My approach was to reinstall the ESP32 board libraries and fastLED libraries and tested by compiling for different boards.   I also found it helpful to install different ESP32 variants for fastLED.

Andreas Spiess youtube channel helped:
https://www.youtube.com/channel/UCu7_D0o48KbfhpEohoP7YSQ/videos

marcmerlin, 2018-12-30:
Glad you got it working. it did look like something was broken with your ESP32 board install (which might not have been your fault).
Do you mind resolving/closing this bug?

marcmerlin, 2019-01-19:
@plankatron  Do you mind resolving/closing this bug?

closed 2019-01-20 by focalintent