FastLED/FastLED #694: State of RGBW support

marcpabst, 2018-12-09:
Hey guys, 

first, I'd like to say how grateful I am for this amazing library. I've been playing with it since 2016 and now I'm starting my first 'real' project. I'm going to use SK6812 LEDs which are available with an additional white channel.  I know that FastLED doesn't support RGBW at the moment. However, I'm aware that @focalintent is planning to do a full rewrite of FastLED that would also support RGBW (or any other combinations of colors for that matter). Any information on when we can expect to use the new version? 

I don't want to pressure anyone (and I'm aware that this is an open source project) but I'm thinking of implementing my own 'hacky' solution - and it would be a pity if a version enabling RGBW support would appear within the next weeks :).

marcmerlin, 2018-12-30:
You could ask on https://plus.google.com/u/0/communities/109127054924227823508 but I haven't heard of RGBW being worked on or being close to release.

focalintent, 2018-12-30:
It is being worked on, just slowly (hopefully in the next few months I'll have a first beta of the library available for some platforms) - it is effectively a re-write of all the low level parts of the library (from the CRGB object on up : )

closed 2019-01-20 by focalintent